AdnanHodzic/displaylink-debian #244: Garbled/Blank screens on Debian buster/sid

CharlesCage, 2019-03-27:
Working through some issues on a Debian buster/sid (KDE Plasma 5.14.5) config.

I'm doing the following to get the outputs to show:
`xrandr --setprovideroutputsource 1 0
xrandr --setproviderouutputsource 2 0`

Then connecting the monitors with:
`xrandr --output DVI-I-2-2 --auto --right-of eDP1
xrandr --output DVI-I-1-1 --auto --right-of DVI-I-2-2
`
I end up with one display full of garbage and the other blank. 

I had the same issue with the Intel xorg.conf.d file, none (removed it), and with the 20-displaylink.conf file as mentioned in troubleshooting guide.

Sorry for the hassle, and appreciate any ideas you might have. Thanks for all your work.

My debug output:
`--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: GT2
Point-LP
VGA: Intel Corporation HD Graphics 520 (rev 07)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x48 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 2 name:Intel
Provider 1: id: 0x1d1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x19c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

CharlesCage, 2019-03-27:
Additionally:

lspci output:

00:00.0 Host bridge: Intel Corporation Skylake Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 520 (rev 07)
00:13.0 Non-VGA unclassified device: Intel Corporation Sunrise Point-LP Integrated Sensor Hub (rev 21)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port (rev f1)
00:1c.1 PCI bridge: Intel Corporation Device 9d11 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-LP LPC Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Multimedia audio controller: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)
02:00.0 Network controller: Intel Corporation Wireless 7265 (rev 61)
03:00.0 Non-Volatile memory controller: Toshiba America Info Systems NVMe Controller (rev 01)

lspci --nnk output:

00:00.0 Host bridge [0600]: Intel Corporation Skylake Host Bridge/DRAM Registers [8086:1904] (rev 08)
        Subsystem: Hewlett-Packard Company Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers [103c:81a1]
        Kernel driver in use: skl_uncore
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 520 [8086:1916] (rev 07)
        Subsystem: Hewlett-Packard Company Skylake GT2 [HD Graphics 520] [103c:8297]
        Kernel driver in use: i915
        Kernel modules: i915
00:13.0 Non-VGA unclassified device [0000]: Intel Corporation Sunrise Point-LP Integrated Sensor Hub [8086:9d35] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP Integrated Sensor Hub [103c:81a1]
        Kernel driver in use: intel_ish_ipc
        Kernel modules: intel_ish_ipc
00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP USB 3.0 xHCI Controller [103c:81a1]
        Kernel driver in use: xhci_hcd
        Kernel modules: xhci_pci
00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Thermal subsystem [8086:9d31] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP Thermal subsystem [103c:81a1]
        Kernel driver in use: intel_pch_thermal
        Kernel modules: intel_pch_thermal
00:16.0 Communication controller [0780]: Intel Corporation Sunrise Point-LP CSME HECI #1 [8086:9d3a] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP CSME HECI [103c:81a1]
        Kernel driver in use: mei_me
        Kernel modules: mei_me
00:1c.0 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port [8086:9d10] (rev f1)
        Kernel driver in use: pcieport
00:1c.1 PCI bridge [0604]: Intel Corporation Device [8086:9d11] (rev f1)
        Kernel driver in use: pcieport
00:1c.4 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 [8086:9d14] (rev f1)
        Kernel driver in use: pcieport
00:1f.0 ISA bridge [0601]: Intel Corporation Sunrise Point-LP LPC Controller [8086:9d48] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP LPC Controller [103c:81a1]
00:1f.2 Memory controller [0580]: Intel Corporation Sunrise Point-LP PMC [8086:9d21] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP PMC [103c:81a1]
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d70] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP HD Audio [103c:81a1]
        Kernel driver in use: snd_hda_intel
        Kernel modules: snd_hda_intel, snd_soc_skl
00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
        Subsystem: Hewlett-Packard Company Sunrise Point-LP SMBus [103c:81a1]
        Kernel driver in use: i801_smbus
        Kernel modules: i2c_i801
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader [10ec:5227] (rev 01)
        Subsystem: Hewlett-Packard Company RTS5227 PCI Express Card Reader [103c:81a1]
        Kernel driver in use: rtsx_pci
        Kernel modules: rtsx_pci
02:00.0 Network controller [0280]: Intel Corporation Wireless 7265 [8086:095a] (rev 61)
        Subsystem: Intel Corporation Dual Band Wireless-AC 7265 [8086:5010]
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
03:00.0 Non-Volatile memory controller [0108]: Toshiba America Info Systems NVMe Controller [1179:010f] (rev 01)
        Subsystem: Toshiba America Info Systems NVMe Controller [1179:0001]
        Kernel driver in use: nvme
        Kernel modules: nvme

CharlesCage, 2019-03-27:
Add'l: Tried the other config:

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

...and still seeing blank screens.

damnhalfling, 2019-05-04:
im having the same problem:

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
Identifier "DisplayLink"
Driver "modesetting"
Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 1 name:Intel
Provider 1: id: 0x1c4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Goober1117, 2019-05-06:
Same here.

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel
Provider 1: id: 0x1dc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x19e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

Thank you for your work on this project.

closed 2020-07-18 by Barabazs