AdnanHodzic/displaylink-debian #249: Fail in debian buster

Khadmium, 2019-04-13:
I have issue with running Xorg after running displaylink installation Script. for convenience i upload log files from journalctl and your script with debug logs. I'm running debian buster with kernel version 4.19.0-4-amd64. According to journalctl logs currently script is breaking nouveau driver (but this can be wrong direction also, i'm not an expert).  Sorry for long journalctl log, to find necessery info grep match against 'Xorg' pattern.
[log_failed_nouveau.log](https://github.com/AdnanHodzic/displaylink-debian/files/3076697/log_failed_nouveau.log)
[log_failed_displaylink.log](https://github.com/AdnanHodzic/displaylink-debian/files/3076699/log_failed_displaylink.log).

Darcidride, 2019-04-15:
Hi,
The fix you're needing is probably in this issue : https://github.com/AdnanHodzic/displaylink-debian/issues/239
Read carefully this comment : https://github.com/AdnanHodzic/displaylink-debian/issues/239#issuecomment-472831580

isdarka, 2019-08-08:
Same problem on Debian Buster 

I had displaylink-debian working ok (Installed with branch nexgen-kernel) in debian strench
I made an upgrade to debian buster and it no longer detects the providers/monitors

**xrandr --listproviders**
```
Providers: number : 0
```

**lsb_release -a**
```
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

```
**uname -r**
```

4.19.0-5-amd64
```

**./displaylink-debian.sh --debug**
```
No protocol specified
Can't open display :0

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

Unit dlm.service could not be found.
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-5-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2
EVDI service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 530
VGA: Intel Corporation HD Graphics 530 (rev 06)
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

** dmesg | grep evdi ** 
```
[    4.060012] evdi: loading out-of-tree module taints kernel.
[    4.060180] evdi: module verification failed: signature and/or required key missing - tainting kernel
[    4.067141] evdi: [I] Initialising logging on level 5
[    4.067144] evdi: [I] Atomic driver:no
[    4.660628] evdi: [D] evdi_add_devices:136 Increasing device count to 1
[    4.660743] evdi: [D] evdi_crtc_init:353 drm_crtc_init: 0
[    4.660788] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[    4.660790] evdi: [D] evdi_detect:88 Painter is disconnected
[    4.660796] evdi: [W] evdi_painter_crtc_state_notify:486 Painter does not exist!
[    4.660876] [drm] Initialized evdi 1.5.1 20180731 for evdi.0 on minor 2
[    4.660877] [drm] evdi: evdi_stats_init
[   11.554221] evdi: [D] evdi_painter_connect:574 (dev=2) Connected with 00000000affeb638
[   11.554224] evdi: [D] evdi_detect:85 (dev=2) Painter is connected
[   11.554288] evdi: [D] evdi_painter_crtc_state_notify:483 (dev=2) Notifying crtc state: 3
```
**systemctl status displaylink-driver.servic**
```
● displaylink-driver.service - DisplayLink Driver Service
   Loaded: loaded (/lib/systemd/system/displaylink-driver.service; static; vendor preset: enabled)
   Active: active (running) since Thu 2019-08-08 16:49:01 CDT; 54s ago
  Process: 763 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/5.2.14 && modprobe evdi) (code=exi
 Main PID: 778 (DisplayLinkMana)
    Tasks: 36 (limit: 4915)
   Memory: 32.1M
   CGroup: /system.slice/displaylink-driver.service
           └─778 /opt/displaylink/DisplayLinkManager
```

closed 2020-07-18 by Barabazs