AdnanHodzic/displaylink-debian #252: Can't open display

carlosalzate74, 2019-04-23:
Dear Friends, 

I have a new Ubuntu Server 18.04 installed on a Toshiba Radius laptop, a usb switch (kvm) between this laptop and the Plugable 3.0. When I run _sudo ./displaylink-debian.sh --debug_ the first message is "Can't Open Display". I am running Ubuntu in text mode, no graphical interface at all.

Any ideas?

Thanks !

AdnanHodzic, 2019-04-23:
You need `sudo apt-get install xorg`, you also might need couple of other X11 related packages.

carlosalzate74, 2019-04-25:
Thanks, I have just installed xorg but same results.

This is the output when I run the debug option:

Can't open display 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-48-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Systems
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.2
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

-------------------------------------------------------------------

carlosalzate74, 2019-05-18:
Any ideas?

marcelolaia, 2020-03-21:
Same problem here. I run the # ./displaylink-debian.sh and got Can't open display. I am try it in debian testing.

kopax, 2020-03-24:
This is no longer maintained, I have stopped using my Asus MB168B+ screen since Debian Stretch and USB-C Laptops. I was only able to use it on Jessie with my previous laptop.

loiselqw, 2020-03-31:
Having the same problem here on bullseye/sid x64 system. 

> Can't open display

When trying to start ./displaylink-debian.sh

Right now I am running 

`DisplayLink Linux Software 5.2.14 install script called: --help
Distribution discovered: Debian GNU/Linux bullseye/sid
`
with  
> /etc/X11/xorg.conf.d/20-displaylink.conf

`Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
`

I need to upgrade displaylink-debian, as lightdm is crashing from time to time when using displaylink-debian 

[https://github.com/DisplayLink/evdi/issues/128](url)

=> I will provide today at evdi project further information.

**At all, how could it be solved to run the script, and not getting the information "Can't open display" anymore?**

Thx in adv

marcelolaia, 2020-03-31:
Em ter, 31 de mar de 2020 08:27, loiselqw <notifications@github.com>
escreveu:

>
> *At all, how could it be solved to run the script, and not getting the
> information "Can't open display" anymore*
>

Hi!

Please, don't run the script in a root terminal. Otherwise, run it in a
user terminal with sudo.

Marcelo

>

Barabazs, 2020-07-19:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs

pyschot49, 2021-07-06:
Hi,

This still persists with my pc, in addition to having a blank gui on my only monitor.
Xrandr, displaylink-debian.sh, all output can't open display.  I can get to TTY but that's it.  If I uninstall the driver all works just fine.

Can someone assit?

timeteam, 2021-10-01:
Hi, likewise, exactly the same problem here under Mint 20.2 Cinnamon.  I think I've tried many of the solutions offered.

can anyone offer suggestions, even trying a different distribution (as a last resort)?  thanks R

R0ad-Runn3r, 2023-02-19:
Hey, are there any news regarding this issue?

I am not able to execute the script with the debug option:
i get the output  "Can't open display"

kind regards

abhignans, 2023-05-27:
I had the same issue and resolved it. 
I got the issue because I was running the script from a terminal using SSH from a different Computer.
This issue was resolved and the script runs when I run directly on the Target Computer by having a display on it.
I believe it also can be an x11 issue.