AdnanHodzic/displaylink-debian #256: (FIXED) Ubuntu 19.04 + Kernel 5.0.0.14 + Intel i915 UHD 620 + Nvidia MX150

mtl010957, 2019-05-02:
Installing with sudo ./displaylink-debian install produces the following setup which freezes on boot when starting up the DisplayLink driver:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: disco
Kernel: 5.0.0-14-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
X11 version: 1.20.4-1ubuntu3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0x1d3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

Deleting the entire 20-displaylink.conf file allows the system to startup but then the mouse exhibits the blocky artifacts and mouse pointer trails reported elsewhere.

The following setup works and eliminates the mouse artifacts:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: disco
Kernel: 5.0.0-14-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
X11 version: 1.20.4-1ubuntu3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0x1d3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

mtl010957, 2019-05-02:
UPDATE: The working configuration only works if the DisplayLink dock is plugged in after X is fully running after login. If plugged in at boot time or before the login the whole system freezes.

mtl010957, 2019-05-06:
Blocky artifacts and mouse pointer trails I get with no 20-displaylink.conf file are the same as reported in #258.

akaytatsu, 2019-05-15:
I have same problem

closed 2020-07-18 by Barabazs