AdnanHodzic/displaylink-debian #257: delay in mouse response and lag

rakishii13, 2019-05-02:
There is a rendering issue where the mouse is flickering and lagging , it also leaves a trail of cursor icons as it is moved on the screen. This started after upgrading to Ubuntu 19.04 and made sure it had nothing to do with that installation, as it is fine when not connected to the displayLink. 
[image]

jonathanelscpt, 2019-06-07:
Am seeing the same in 18.10.

[image]

Crashdummyy, 2019-06-24:
did you try deleting the displaylink.conf ?
I use displaylink on my MSI GS70 with the same CPU, where it works flawlessly

closed 2020-07-18 by Barabazs