AdnanHodzic/displaylink-debian #261: Building EVDI kernel module with DKMS fails on ubuntu 18.04 with kernel 5.1.0-050100-generic

vthuongt, 2019-05-07:
When I try to install the driver the `/displaylink-debian.sh` will fail at:

> Installing driver version: 5.1
> 
> DisplayLink Linux Software 5.1.26 install script called: install
> Distribution discovered: Ubuntu 18.04.2 LTS
> Installing
> Configuring EVDI DKMS module
> Registering EVDI kernel module with DKMS
> Building EVDI kernel module with DKMS
> ERROR (code 3): Failed to build evdi/5.1.26. Consult /var/lib/dkms/evdi/5.1.26/build/make.log for details..

The log file contains:
<details> 
> DKMS make.log for evdi-5.1.26 for kernel 5.1.0-050100-generic (x86_64)
Di 7. Mai 11:07:42 CEST 2019
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/5.1.26/build SRCROOT=/var/lib/dkms/evdi/5.1.26/build CONFIG_MODULE_SIG= -C /lib/modules/5.1.0-050100-generic/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.1.0-050100-generic'
Makefile:205: ================= WARNING ================
Makefile:206: 'SUBDIRS' will be removed after Linux 5.3
Makefile:207: Please use 'M=' or 'KBUILD_EXTMOD' instead
Makefile:208: ==========================================
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/lib/dkms/evdi/5.1.26/build/.tmp_versions ; rm -f /var/lib/dkms/evdi/5.1.26/build/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/var/lib/dkms/evdi/5.1.26/build
  gcc -Wp,-MD,/var/lib/dkms/evdi/5.1.26/build/.evdi_drv.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_drv"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/5.1.26/build/evdi_drv.o /var/lib/dkms/evdi/5.1.26/build/evdi_drv.c
  gcc -Wp,-MD,/var/lib/dkms/evdi/5.1.26/build/.evdi_modeset.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_modeset"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/5.1.26/build/evdi_modeset.o /var/lib/dkms/evdi/5.1.26/build/evdi_modeset.c
  gcc -Wp,-MD,/var/lib/dkms/evdi/5.1.26/build/.evdi_connector.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-int-in-bool-context -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_connector"' -DKBUILD_MODNAME='"evdi"' -c -o /var/lib/dkms/evdi/5.1.26/build/evdi_connector.o /var/lib/dkms/evdi/5.1.26/build/evdi_connector.c
  gcc -Wp,-MD,/var/lib/dkms/evdi/5.1.26/build/.evdi_encoder.o.d  -nostdinc -isystem /usr/lib/gcc/x86_64-linux-gnu/7/include -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Wno-format-security -std=gnu89 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -DCONFIG_X86_X32_ABI -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -DCONFIG_AS_SSSE3=1 -DCONFIG_AS_AVX=1 -DCONFIG_AS_AVX2=1 -DCONFIG_AS_AVX512=1 -DCONFIG_AS_SHA1_NI=1 -DCONFIG_AS_SHA256_NI=1 -Wno-sign-compare -fno-asynchronous-unwind-t [... cut ...]
Makefile:1571: recipe for target '_module_/var/lib/dkms/evdi/5.1.26/build' failed
make[1]: *** [_module_/var/lib/dkms/evdi/5.1.26/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.1.0-050100-generic'
Makefile:22: recipe for target 'all' failed
make: *** [all] Error 2

</details> 

I checked the paths `/usr/src/linux-headers-5.1.0-050100-generic/include/generated/autoconf.h` and  `/usr/src/linux-headers-5.1.0-050100-generic/include/config/auto.conf` but both existed. So I don't know why this error appears. How can I fix this error? Thank you in advance!

zash1958, 2019-05-16:
Hi,
is there an "howto" for updateing EVDI? I downloaded the sources from github and can compile it. But what to do with the resulting modules and files?

TIA

Uwe

julianofcald, 2019-05-23:
Hi,

I've created a modified version of DisplayLink driver 5.1.26 with the new included evdi v1.6.1 which worked well on elementary OS Juno5 (ubuntu LTS 18.04.02 based) and kernel v5.1.4 from ukuu. 

I renamed it to v5.1.27 and attached a working installer to solve our problem. 

You can use the installer exactly as explained by https://support.displaylink.com/knowledgebase/articles/679060

by running as super user displaylink-installer.sh file after extracting the tar.gz file:

`sudo ./displaylink-installer.sh --install`

Best regards!

[displaylink-driver-5.1.27.tar.gz](https://github.com/AdnanHodzic/displaylink-debian/files/3210061/displaylink-driver-5.1.27.tar.gz)

zash1958, 2019-05-23:
Thanks a LOT!!!!!

Installs and works also perfectly under Debian-SID with following kernel:
5.1.3-towo.2-siduction-amd64

Both external displays on my notebook were detected as expected.

Thanks for Your work

Luc45, 2019-07-01:
I can't get it to work either way.

Using `sudo ./displaylink-debian.sh --install`:

```
➜  displaylink-debian git:(master) sudo ./displaylink-debian.sh --install

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.1.9-050109-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1304
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2019-07-01 16:39:48--  https://www.displaylink.com/downloads/file?id=1304
Resolving www.displaylink.com (www.displaylink.com)... 34.246.147.251, 18.203.109.3, 54.77.143.153, ...
Connecting to www.displaylink.com (www.displaylink.com)|34.246.147.251|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1304_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1562011070&Signature=r2oaiHm0VD9H1ObyKIwzFmPv1EY%3D [following]
--2019-07-01 16:39:49--  http://assets.displaylink.com/live/downloads/software/f1304_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1562011070&Signature=r2oaiHm0VD9H1ObyKIwzFmPv1EY%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.36.26
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.36.26|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12102697 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.1.zip’

DisplayLink_Ubuntu_5.1.zip           6%[==>                                                           ] 721,20K  97,5KB/s    eta 1m 4DisplayLink_Ubuntu_5.1.zip        100%[==========================================================>]  11,54M  27,3KB/s    in 4m 52s  

2019-07-01 16:44:42 (40,4 KB/s) - ‘DisplayLink_Ubuntu_5.1.zip’ saved [12102697/12102697]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.1.zip
  inflating: 5.1/displaylink-driver-5.1.26.run  
  inflating: 5.1/LICENSE             
Creating directory displaylink-driver-5.1.26
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.1.26  100%  

-------------------------------------------------------------------

Installing driver version: 5.1

DisplayLink Linux Software 5.1.26 install script called: install
Distribution discovered: Ubuntu 18.04.2 LTS
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS

==== \/ HERE \/  ====
ERROR (code 3): Failed to build evdi/5.1.26. Consult /var/lib/dkms/evdi/5.1.26/build/make.log for details..
==== /\ HERE /\ ====
[... 71 lines cut ...]
Distribution discovered: Ubuntu 18.04.2 LTS
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
Installing EVDI kernel module to kernel tree
EVDI kernel module built successfully
Installing x64-ubuntu-1604/DisplayLinkManager
Installing libraries
chmod: cannot access '/opt/displaylink/DisplayLinkManager': No such file or directory
chmod: cannot access '/opt/displaylink/libevdi.so': No such file or directory
chmod: cannot operate on dangling symlink '/opt/displaylink/libusb-1.0.so'
chmod: cannot operate on dangling symlink '/opt/displaylink/libusb-1.0.so.0'
Installing firmware packages
mv: cannot stat '*.spkg': No such file or directory
chmod: cannot access '/opt/displaylink/*.spkg': No such file or directory
Installing licence file
Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu
Installation complete!
```

As you can see, it passes the EVDI kernel module step but throws some warnings for some files not being found.

julianofcald, 2019-07-02:
@Luc45 ,

Are you using 64bit or 32bit Ubuntu version? My 64bit eOS setup do not required those files.
Double check it, because the provided files are only for x86 and arm. The x64-ubuntu-1604 folder is empty.

Luc45, 2019-07-02:
@julianofcald 

The first time I saw those "missing file" warnings, I copied the contents from the x86 folder you provided:

`sudo cp displaylink-driver-5.1.27/x86-ubuntu-1604 /opt/displaylink`

Upon seeing the warnings were gone but the adapter still didn't work, I uninstalled it again, downloaded the driver from DisplayLink, copied the x64 folder from there:

`sudo cp displaylink-driver-5.1.26/x64-ubuntu-1604 /opt/displaylink`

And installed yours (5.1.27, with EVDI 1.6.1). It installed fine again, but still, the adapter wasn't working.

Does yours work? Can you please run `sudo ls -la /opt/displaylink` and let me know if there's anything in there?

Luc45, 2019-07-02:
Okay, so I got this working - sort of.

I downloaded a fresh version of 5.1.26 from DisplayLink drivers and got the EVDI 1.6.1 .tar.gz from the file you shared, @julianofcald. Now, I have to follow a very specific step to get this working:

- With the adapter unplugged I turn on the computer
- I start the service manually with `sudo systemctl start displaylink-driver`
- I plug in the adapter

If I plug it out, the service becomes inactive. If I start it again and connect the adapter again, it doesn't work - it's a one time thing, only the first time it is plugged in or I have to reboot.

Other than that, there are some artifacts around the mouse on the DisplayLink monitor and mouse is very lagged. I use GTX 1070 with Nvidia drivers, but that's a subject for another issue...

Luc45, 2019-07-02:
So, I have uninstalled NVIDIA Drivers, and came back to Noveau drivers. Ugraded kernel version from 5.1.9 to 5.1.15

Adapter now works on boot, without me having to do the steps outlined in the previous comments. No artifacts shows up. However, if I make any changes in the displays, such as closing the laptop lid (which turns off 1 display), or changing the orders of the displays, the artifacts around the mouse, such as pieces of the screen, starts to show up. Weird. I have no idea how to proceed.

Luc45, 2019-07-03:
So, more tips for anyone who cares:

I managed to solve the log-in/log-out artifacts with these:
- https://support.displaylink.com/knowledgebase/articles/1859356-x-server-crashes-on-nvidia-systems-when-displaylin
- https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only

The reason I did those is because /opt/displaylink/DisplayLinkManager was using a LOT of CPU, specially when moving my mouse around (I know, right?)

Seems to have helped it, even though it still uses more CPU than I would like.

julianofcald, 2019-07-06:
Hey @Luc45 ,

Thank you so much for sharing those links with us.
You solved an old problem that i have here. Before your tip, my dell displaylink dock station only displayed image if i booted with the USB cable plugged in before boot. Additionally, before suspending the notebook, the displaylink device must be removed in order to continue working after wake up. Now, i don't have the problem anymore.

However, i don't know if it's because in using eOS, but i don't have the /etc/modules listed in one of your links. So, my workaround was to:

`sudo nano /usr/share/X11/xorg.conf.d/21-displaylink.conf`
paste:
`Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection`
saved the file. (exactly as said in the first article)

and then:
`sudo nano /etc/modules-load.d/modules.conf`
added: 
`evdi`
saved the file.

and then:
`sudo nano /etc/modprobe.d/evdi.conf`
added:
`options evdi initial_device_count=2`
because i have 2 extra screens
saved the file.

Rebooted.

and then, like magic, it worked. 
Thank you so much for sharing those links.

saikrishnagaddipati, 2019-07-08:
@Luc45  i follwed the steps in this thread but my display link disconnects/hungs intermittently sometimes. can you please give details steps/commands on how you installed displaylink.
My ubuntu version is 18.05, kernel_version: 5.1.9

Luc45, 2019-07-09:
@julianofcald 
I'm glad that I could be of help!

@saikrishnagaddipati 
I didn't face this issue, so I don't know, but I suspect of some kind of USB power management thing. (https://askubuntu.com/questions/185274/how-can-i-disable-usb-autosuspend-for-a-specific-device)

closed 2020-07-18 by Barabazs