AdnanHodzic/displaylink-debian #262: Display not detected - USB Dell dock on Dell XPS - Debian testing/sid

daks, 2019-05-10:
Hello,

I have a Dell XPS 13 laptop with a Dell USB dock. The laptop have an integrated mini-DP port _(which works well)_ and the dock 2 HDMI ports and a DP one.
I tried several times to make my OS detect those available on the dock to finally being able to have two external monitors... without success.

```
sudo ./displaylink-debian.sh --debug

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-4-amd64

---------------- DisplayLink info ----------------

Driver version: 1.1.62
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel
Provider 1: id: 0x1cf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

After installing and rebooting, I see new providers
```
xrandr --listproviders
Providers: number : 3
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel
Provider 1: id: 0x1cf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

but I can't see them in `arandr` which only displays my default eDP1 internal display.

Watching `dmesg`, I see the USB displaylink hub appear when I plug it, and can see it using `lsusb`
```
lsusb|grep Display
Bus 003 Device 012: ID 17e9:436e DisplayLink 
Bus 003 Device 013: ID 17e9:436e DisplayLink 
```
I can look at `/var/log/displaylink/DisplayLinkManager.log` but I only see binary garbage, nothing useful.

Any help would be appreciated.

davewuvswaffles, 2019-10-09:
I'm having the same issue - were you ever able to get through this?

At first I thought it was an issue with Wayland but am now seeing the same behavior in X11.

Debian Sid 5.2.0-3-amd64

daks, 2019-10-09:
No, I didn't tested a lot more, but it never worked.

nobodyman1, 2019-11-24:
Did you try the following commands?

```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
```
After calling `xrandr` (without parameters) it shows something like `DVI-I-1-1 connected`.
To activate the display use:
`xrandr --output DVI-I-1-1 --auto`
But there is still one problem. This works after you are logged in.

closed 2020-07-18 by Barabazs