AdnanHodzic/displaylink-debian #265: Choppy image on Kubuntu 19.04 with Ryzen 2500U

nicoco007, 2019-05-15:
I recently got a new laptop and finally got around to installing Kubuntu 19.04 on it. I tried running this script as I did on my old laptop (on which it worked nicely) and DisplayLink is able to find and show an image on my external monitor (an ASUS MB169B+, 1080p via USB 3.0) but unfortunately the image is very choppy, even when just the cursor is moving. The laptop screen itself stays smooth and there isn't an obvious reduction in peformance when the monitor is plugged in. CPU usage stays relatively low (reaches ~50% on one core when moving a window around on the external monitor).

Things I've tried:

* Update kernel to version 5.1.2 - driver 5.1.26 fails to compile
* Downgrade kernel to 4.19.43 and 4.18.20 - problem remains
* Downgrade driver to 4.4.24 - compiles only against kernel 4.18.20 and problem remains
* Follow what was done in #180 - DisplayLink service no longer starts and the xrandr provider disappears

Debug information:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: disco
Kernel: 5.0.0-15-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: [Radeon
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c4)
VGA (3D): 
X11 version: 1.20.4-1ubuntu3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x99 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 2 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0
Provider 1: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

Provider 2 appeared after I tried a few random things to fix the problem; it doesn't seem to affect anything at this point in time but it might be a good idea to figure out how to get rid of it.

Not sure if this is useful, but here's the output from the commands asked in #180:
```
$ cat /etc/X11/xorg.conf.d/20-displaylink.conf
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

$ lspci | grep -i vga
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c4)
03:00.7 Non-VGA unclassified device: Advanced Micro Devices, Inc. [AMD] Device 15e6

$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
amdgpu
i2c_amd_mp2

$ lspci -nnk | grep -i vga -A3|grep Subsystem|cut -d" " -f5
[Radeon
Inc.
```

I'm guessing it's something to do with the xorg configuration, but I have no idea where to start. Any ideas?

Crashdummyy, 2019-06-24:
Hey Nico, any Solutions yet?
I am facing the Same Issue as you Running a Hp ENvy x360 with an Ryzen 2700u ( Vega 10 )

```bash
--------------- Linux system info ----------------

Distro: Ubuntu
Release: disco
Kernel: 5.0.0-17-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Ridge
[103c:8497]
SATA
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)
VGA (3D): 
X11 version: 1.20.4-1ubuntu3
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 2 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x5dc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x58a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

```

I couldn't get an Image to begin with, so I removed the .conf as suggested.

```sh
$ lspci | grep -i vega
04:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Raven Ridge [Radeon Vega Series / Radeon Vega Mobile Series] (rev c3)

$ lspci -nnk | grep -i vga -A3 | grep 'in use'|cut -d":" -f2|sed 's/ //g'
amdgpu

```

nicoco007, 2019-06-26:
Not really, but I've come to the conclusion that it might be an issue with EVDI itself since I couldn't get it to work properly on Ubuntu 18.04, so I [created another issue over on their tracker](https://github.com/DisplayLink/evdi/issues/164).

Crashdummyy, 2019-06-26:
When I use my Laptopscreen as well, the Lag is slightly reduced.
Is that the same for you ?

Crashdummyy, 2019-06-26:
@nicoco007 Thanks a lot, I see you have the same Problems as I do ( like immense stuttering if only using the External Displays )

Crashdummyy, 2019-06-27:
Hey @nicoco007 i don't know how this would work out on your end.
But when I use Wayland instead of X the stuttering is (almost) completely gone ;)

nicoco007, 2019-06-27:
What's your setup? I'm just getting a black screen or straight up crashes when trying to use Wayland.

Crashdummyy, 2019-06-27:
That's strange.
I'm just using the latest 19.04 with Linux Kernel 5.0.0.17.20

It uses GDM3 where you can turn on Wayland on the LogIn screen ( there is a Cog somewhere... )

nicoco007, 2019-06-27:
Weird. I don't get a cog on `5.0.0-19-generic`, and I doubt change from 17 to 19 made it disappear...

nicoco007, 2019-06-27:
Nevermind, I must have done something wrong the first time because I just got it to work. Kind of weird it works so well since I'm pretty sure they don't actually support Wayland yet...

nicoco007, 2019-06-27:
Unfortunately that doesn't work on Kubuntu, most likely because KDE Plasma doesn't fully support Wayland yet :(

Crashdummyy, 2019-06-28:
I wouldn't use Gdm3 either if it were not for the Touchscreen Experience :laughing: 
No problems with displaylink on i3 :)

legion49f, 2019-10-07:
I switched to Wayland and the framerate is around 50fps now

Barabazs, 2020-07-18:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-07-18 by Barabazs

dkebler, 2021-09-16:
reopening this as it seems closely related.  I had been using my displaylink device for couple years with an intel I5 (integrated gpu) without issue.
Just switched over to a ryzen 7 3750u machine

The issue is that the display is lagging/slow which is very apparent when playing a video.  When I unplug the displaylink device it goes away, plug back in it returns.

this default conf file (set by install) is 

```
Section "OutputClass"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option  "AccelMethod" "none"
EndSection
```

and I have also tried with no conf file but the issue remains.
I run linux mint 20 and it's not possible to use wayland so that is not a solution for me.
maybe I need accelleration of some kind?
Any suggestions.

here is my debug output.  That's mint 20 which uses focal 20.04.

```
--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana  (focal 20.04)
Kernel: 5.4.0-84-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.2
1.9.1
DisplayLink service status: 
EVDI service version: 1.9.1

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: Inc.
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c1)
VGA (3D): 
X11 version: 1.20.11-1ubuntu1~20.04.2
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 
note: default mentioned above did not work either

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x57 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 4 name:AMD Radeon(TM) Vega 10 Graphics @ pci:0000:05:00.0
Provider 1: id: 0x100 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xde cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xbc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x9a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```