AdnanHodzic/displaylink-debian #266: Debain freez when turning on external display

ddkovi, 2019-05-19:
Dear Adnan Hodzic,

I have some problem with a displaylink on debian.
I think I miss something, but I don't know what.

I have one [Lenovo Thinkpad T470](https://www.lenovo.com/gb/en/laptops/thinkpad/t-series/ThinkPad-T470/p/22TP2TT4700) with i5-7300U cpu and integrated Intel® HD Graphics 620 gpu, and I try combinate it with a [Lenovo Thinkpad USB 3.0 Pro Dock](https://www.lenovo.com/gb/en/accessories-and-monitors/docking/universal-cable-docks-usb/ThinkPad-USB-3-0-Pro-Dock/p/40A70045UK).

I made a fresh install [Debian 9 non-free](https://cdimage.debian.org/cdimage/unofficial/non-free/cd-including-firmware/current/amd64/iso-cd/firmware-9.8.0-amd64-netinst.iso) (non-free because of wifi drivers, with Cinnamon DE), after change the BIOS setings:
**Secure Boot** - Disabled
**UEFI/Legacy Boot** - Legacy Only

After install the displaylink-debian I have a following debug information:
[debug after install.txt](https://github.com/AdnanHodzic/displaylink-debian/files/3088108/debug.after.install.txt)
When I run `echo $DESKTOP_SESSION` and `echo $XDG_SESSION_TYPE` in terminal I give the following output: `lightdm-xsession` `x11`. So I think I am use X server, not Wayland.

The deafult content of **20-displaylink.conf** file under /ect/X11/xorg.conf.d/ is:
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
With it the `xrandr --listproviders` command only give one provider.

I tried the following configurations in the **20-displaylink.conf** file:
- With these configurations the system wont start:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
```
Section "OutputClass"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    MatchDriver "evdi"
    Option      "PageFlip" "false"
    Option      "VSync" "false"
EndSection
```
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "evdi"
    Option      "PageFlip" "false"
    Option      "VSync" "false"
EndSection
```
- With the following configurations the system start, and after `xrandr --setprovideroutputsource 1 0` command the display manager see the external monitors, but when I activate it, the system freeze:
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
    Option "VSync" "false"
EndSection
```
- the following options result: sytem start, `xrandr --listproviders ` only give one provider:
```
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
#	Option "VSync" "false"
EndSection
```
```
Section "Device"
	Identifier "Intel"
	Driver "intel"
#	Option "AccelMethod" "uxa"
	Option "VSync" "false"
EndSection
```
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
    Option "VSync" "false"
EndSection
```
- and with this configuration:
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
    Option      "VSync" "false"
    Option      "AccelMethod" "uxa"
EndSection
```
I gave error on running `xrandr --setprovideroutputsource 1 0` command:

> X Error of failed request:  BadValue (integer parameter out of range for operation)
>   Major opcode of failed request:  140 (RANDR)
>   Minor opcode of failed request:  35 (RRSetProviderOutputSource)
>   Value in failed request:  0x69
>   Serial number of failed request:  16
>   Current serial number in output stream:  17

When the system freez after turning on the external display, only the power off (holding the power button) works, even the Caps Lock not reacting.

I made a [Linux Mint](https://quantum-mirror.hu/mirrors/linuxmint/iso/stable/19.1/linuxmint-19.1-cinnamon-64bit.iso) install (it is based on Ubuntu), after installed the [DisplayLink Ubuntu](https://www.displaylink.com/downloads/file?id=1304) driver it is work perfectly, so the probleam not in the hardvares. I think I have wrong configuration in the 20-displaylink.conf file, but I can't found the good one.

Thank you for your help.
Norbert Kovács

beshad, 2019-06-21:
i am now having the exact problem. debian freezes as soon as the second monitor connects

blkandyllw, 2019-06-21:
Same - Asus K55VM - debian 9.9

damnhalfling, 2019-07-10:
I have the same problem, any solution?

ddkovi, 2019-07-10:
I tested the following configurations, but none of them wokring. :(

It is a same: freez
Section "Device"
	Identifier "Intel Graphics"
	Driver "modesetting"
	Option "PageFlip" "false"
EndSection

Whit this the system wont start:
```
Section "Device"
	Identifier "Intel Graphics"
	Driver "i915"
	Option "PageFlip" "false"
EndSection
```

blkandyllw, 2019-08-23:
I solved my problem by upgrading Debian 9 to 10

beshad, 2019-08-23:
Yes this was solved for me too when updated to Buster.

ddkovi, 2020-04-06:
The problem is solved for me too by upgrading Deabin 9 to 10.

closed 2020-04-06 by ddkovi