AdnanHodzic/displaylink-debian #269: Not working with Amazon Docking Station

tcov189, 2019-06-04:
After installing and following the post-install scripts, I cannot get my external monitors to work. 
I am trying to use my AmazonBasics USB 3.0 Universal Laptop Dual Monitor Docking Station to connect my other two monitors.
Here is the debug output
```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.18.0-21-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.1-3ubuntu2.1~18.04.1
X11 configs:

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting

```

hcallaway, 2019-06-10:
I am having a similar output, everything under DisplayLink Info is blank, also on Amazon Basics dock.

tcov189, 2019-06-20:
So my issue was that safe boot was still enabled. I thought that I had disabled it, but it turns out that I am dumb.

@hcallaway I would double check that safe boot is disabled in your BIOS.

closed 2019-06-20 by tcov189