AdnanHodzic/displaylink-debian #270: debian buster and dell xps 13 9380

MNatiello, 2019-06-06:
Hallo

Displaylink  behaves erratically. It detects the monitor but fails to open a connection.
My displaymanager keeps asking for configuration, but the screen of the second monitor
is always black.
I wonder if this is related to the fact that for a given resolution the laptop and the monitor have different refresh rates. 
Any question or suggestion is welcome.
M. Natiello

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-5-amd64

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake)
VGA (3D): 
X11 version: 1.20.3-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0x492 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

MNatiello, 2019-06-07:
Addendum
Eventually, I get it to work after a number of trials from the device and my answers.
I could mirror the screens at 1680x1050 59.9Hz (for reasons unknown), then
de-mirror them, set the laptop display to unused and finally changing the resolution
of the monitor to 3840x2160

jgarbora, 2019-07-09:
HI @MNatiello 
Trying to make this work too. 
I got Debian 10 too, Kernel 4.19.0 ... I only installed evdi-dkms and libevdi0, 
should I run this script ?

MNatiello, 2019-07-09:
Update. Only one of the three ports of the laptop provides the appropriate capabilities for handling the
external screen with full resolution, namely the one on the right side. The working configuration is to
* Connect the displaylink to that port
* Connect the external screen to the displaylink with the DP-cable
* run the following script or equivalent after connection
#!/bin/bash
#
sleep 5
A="$(xrandr --listproviders|grep Providers | gawk '{ print $4 }')"
if [ $A -eq 2 ]
then 
 xrandr --setprovideroutputsource 1 0
xrandr --output DVI-I-1-1 --primary --same-as eDP-1
 xrandr --output eDP-1 --off
fi
After a few seconds, the correct image shows up on the external screen.

closed 2019-07-09 by MNatiello