AdnanHodzic/displaylink-debian #272: displaylink-debian will not install on Deepin 4.15.0-30

htjunky, 2019-06-14:
I am trying to install displaylink-debian on my new Deepin install and it can't find the correct headers.  It is looking for linux-source-deepin and can't find it.  I ran the install with the --debug command:

------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Deepin
Release: stable
Kernel: 4.15.0-30deepin-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Integrated
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.19.6-2deepin
X11 configs: /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

During the install I get this error message:

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.15.0-30deepin-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: package 'linux-source-deepin' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
linux-source-deepin not found! Install? [Y/n] y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-source-deepin
linux-source-deepin installation failed.  Aborting.

Any help on this would be greatly appreciated.

Thanks

trenoncourt, 2019-07-09:
Hi, I have the same error.
Have you find a solution ? 
thx

adamth0mps0n, 2019-07-29:
I have the same error..

Tset-Noitamotua, 2019-08-12:
I am sad about having to join this club :sob: 

[image]

```
cat /etc/issue
Deepin GNU/Linux 15.11 \n \l
```

adamth0mps0n, 2019-08-12:
My current workaround is to install the deepin desktop on a Ubuntu base.

Tset-Noitamotua, 2019-08-13:
> My current workaround is to install the deepin desktop on a Ubuntu base.

How is the UX / "look and feel" with that? Is it close to native Deepin?

adamth0mps0n, 2019-08-13:
It looks and feels the same to me. Mind you I only used native Deepin for about 30 minutes and went to Ubuntu when I couldn't get display-link to work.  The only issues that I have found is that some apps from the deepin store don't work properly on the Ubuntu base, so stick with the native Ubuntu versions and everything is fine. I use Synaptic for my apps, no problems at all.

Tset-Noitamotua, 2019-08-25:
@adamth0mps0n  Thank you for feedback. It sounds like it's worth a try.  But first I have figured out and setup a backup/restore routine of my productive Deepin environment. As much as I love Deepin sometimes one have to find workarounds for things that are supported on Ubuntu out of the box.

To give some feedback to related issue: some how I managed it to get my "**Thinkpad Hybrid USB-C with USB-A**" Dock working with two external Monitors connected to it. But it was clunky and did not survived any reboot. After applying different suggested solutions for some other issues I had with a X1 Carbon 6th (XC16) as well - not just the DisplayLink issue - I finally killed my system and had to do a fresh install :sob: So I guess I won't try that any time soon again.

But I'd be happy to hear from somebody who got a X1 Carbon 6th working with two monitors and any Thunderbold Dock (don't have to be by Lenovo).

closed 2020-07-19 by Barabazs