AdnanHodzic/displaylink-debian #274: No additional providers detected

nderr, 2019-06-18:
Debug information from each of the three possibilities for /etc/X11/xorg.conf.d/20-displaylink.conf are below:
[debug_none.txt](https://github.com/AdnanHodzic/displaylink-debian/files/3303305/debug_none.txt)
[debug_device.txt](https://github.com/AdnanHodzic/displaylink-debian/files/3303306/debug_device.txt)
[debug_output_class.txt](https://github.com/AdnanHodzic/displaylink-debian/files/3303307/debug_output_class.txt)

All three (aside from the section reporting on the config file's contents) are the same, and in each case a single provider is detected. However, in the case with the "Device" config file

Doloros, 2019-06-25:
up here ! same here !

nderr, 2019-07-29:
The new stock DisplayLink driver (5.2.14) fixes this. The DA100 now works with a 24in Lenovo monitor with no configuration needed after installation.

closed 2020-07-18 by Barabazs