AdnanHodzic/displaylink-debian #275: Add support fort deepin 15.10.1

mdemauroy, 2019-06-20:
Would be thrilled if 15.10.1 could be listed in supported versions :D

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: Aucun fichier ou dossier de ce type
--------------- Linux system info ----------------

Distro: Deepin
Release: stable
Kernel: 4.15.0-29deepin-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.0-2deepin
X11 configs: /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/imwheel/startup.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

AdnanHodzic, 2019-06-21:
As part of last commit 660b793 I've extended support to Deeping release you requested. Try performing uninstall followed by install.

closed 2019-06-21 by AdnanHodzic

mdemauroy, 2019-06-21:
Thanks ! Install went smoothly :D

I can't test multi monitors right now, but I will get back to you as soon as I can.

That was super fast, thanks a lot !!

AdnanHodzic, 2019-06-21:
Most welcome, please consider [donating to the project](https://github.com/AdnanHodzic/displaylink-debian#donate-paypal)!

commit 660b7932 (2019-06-21): added support for deeping stable (close #275)