AdnanHodzic/displaylink-debian #287: Unable to find linux-source-deepin package

trenoncourt, 2019-07-12:
hI, i have a problem while trying to install, seems like one package doesn't exist: 'linux-source-deepin'

Here is what i get when i try to install :
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.15.0-30deepin-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: le paquet « linux-source-deepin » n'est pas installé et aucune information n'est disponible
Utilisez dpkg --info (= dpkg-deb --info) pour examiner les fichiers
archives, et dpkg --contents (= dpkg-deb --contents) pour afficher leur
contenu.
linux-source-deepin not found! Install? [Y/n] Y
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
E: Impossible de trouver le paquet linux-source-deepin
linux-source-deepin installation failed.  Aborting.

-------

Here is debug logs:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Deepin
Release: stable
Kernel: 4.15.0-30deepin-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 5917 (rev 07)
VGA (3D): 
X11 version: 1.19.6-3deepin
X11 configs: /etc/X11/imwheel/startup.conf /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/xorg.conf.d/75-wacom.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x75 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Thanks for your help

xuebaofeng, 2020-03-09:
same issue, thanks

lucianefalcao, 2020-03-20:
I`m getting the same issue.

Barabazs, 2020-07-19:
Closing this issue due to inactivity of OP.
Please reopen this issue if it still persists.
@lucianefalcao @xuebaofeng @Tset-Noitamotua  Please open a new issue if you still experience problems.

closed 2020-07-19 by Barabazs