AdnanHodzic/displaylink-debian #290: Intermittent connection breaks while running

iandeed, 2019-07-21:
Using the lenovo thinkpad X1 extreme running with two attached monitors, the laptop appears to lose connection with the dock every 5-10 minutes. This flicks everything off and restores its state back to the way it was around 80% of the time.

I've tried swapping between two USB cables. The issue happens with Thunderbolt bios assist both disabled and enabled. Secure boot and Thunderbolt security are completely disabled.  Debug information below:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.18.0-25-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: GTX
VGA: NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1ubuntu3~18.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x21e cap: 0x1, Source Output crtcs: 4 outputs: 4 associated providers: 2 name:NVIDIA-0
Provider 1: id: 0x84 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

derrickmehaffy, 2019-07-27:
I noticed this same thing as well

iandeed, 2019-07-30:
Update: After further investigation, I'm reasonably convinced the issue is within the laptop itself. It appears the leptop cuts the connection to the thunderbolt port periodically, more often when under heavy load.

The issue has been confirmed on Thinkpad X1 Carbon and X1 extreme with multiple devices, including USB-C chargers and multiple operating systems.

Closing issue as unrelated.

closed 2019-07-30 by iandeed