AdnanHodzic/displaylink-debian #293: Debien Buster - Intel BlackScreen - Modesetting Mouse Flicker and Lag

rrajkomar, 2019-08-20:
Hi,
I've recently upgraded my debian laptop to the Buster version and was faced with several issues, the first of  which was that the default displaylink driver did not work with the new kernel.

This was fixed thanks to your solution.

I'm now faced with two problems : 
 * if I set my /etc/X11/xorg.conf.d/20-intel.conf to the following, I get a blank screen upon running the xrandr commands to launch my external monitor

```
Section "Device"
	Identifier "Intel"
	Driver "intel"
EndSection
```

* if I set my /etc/X11/xorg.conf.d/20-intel.conf to the following, I get working screen that flickers from time to time and has lots of mouse lags(possibly refresh issues but changing the refresh rate did not change anything) upon running the xrandr commands to launch my external monitor

```
Section "Device"
	Identifier "Intel"
	Driver "modesetting"
EndSection
```

Here's the xrand commands used :
* intel
`xrandr --setprovideroutputsource 1 0 --output HDMI1 --right-of eDP1 --output DVI-I-1-1 --left-of eDP1 --mode 1920x1080 --rate 60 --auto`

* modesetting 
`xrandr --setprovideroutputsource 1 0 --output HDMI-1 --right-of eDP-1 --output DVI-I-1-1 --left-of eDP-1 --mode 1920x1080 --rate 60 --auto`

Any help would be greatly appreciated.

If possible using the intel driver as the only issue there seems the lack of display (though the monitor is detected and seems to be working minus the blank screen)

Otherwise by helping me solv the flickering and lag issues with the modesetting driver

rrajkomar, 2019-08-26:
Note : Tried to rerun the same process as above this morning. Now the modesetting option no longer works.
When I try to liste the providers using `xrandr --listproviders`, I get two lines with modesetting
But when I try to run the `xrandr --setprovideroutputsource 1 0` command I get an our of range error, the same mentionned here : https://github.com/AdnanHodzic/displaylink-debian/issues/160

Now I'm stuck with the intel option only which causes my monitor to go blank...

rrajkomar, 2019-08-26:
P.S. : debug info : 

```

rrajkomar, 2019-09-04:
Hello,
@AdnanHodzic any ideas?

BramFr, 2019-09-23:
I have the same problem with mouse flicker/lag. This fixed my issue: 

https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only

rrajkomar, 2019-10-09:
Upgraded to latest master version just in case. same issue still applies : I get a black screen with only the mouse pointer working.
Windows look like they'e on the black screen but are not actually displayed.

rrajkomar, 2019-12-17:
@AdnanHodzic up

rrajkomar, 2020-02-17:
Updated to latest master version, still not working (Note : I don't even have the mouse pointer anymore).
I'm still getting a black empty screen. And it looks like I'm not the only from all the other issue I saw regarding Debian buster.
@AdnanHodzic Do you have any ideas as to why the screen is not showing any windows please ?

nis267, 2020-05-02:
Hello, @rrajkomar 

- Try this in config in your /etc/X11/xorg.conf.d/20-displaylink.conf file and reboot

`Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection`

rrajkomar, 2020-05-03:
@nis267 Thanks for your input but I already tried this (see https://github.com/AdnanHodzic/displaylink-debian/issues/293#issuecomment-524834073)

It doesn't work properly

nis267, 2020-05-04:
@rrajkomar, 

- In your debug output i saw that you have multiple config files can move or remove them all, except the /etc/X11/xorg.conf.d/20-displaylink.conf one if it already exist or create a new one.

- And if you have /etc/X11/xorg.conf file rename it to xorg.conf.org.bak.

- In my debug output you can see that I have only one conf file. My Intel card and the displaylink use both the modesetting driver. I think that when you have an intel driver and modesetting driver both will not work together that's maybe why it's not working for you.

'''
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x8e cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
'''

Bonne journée

rrajkomar, 2020-05-04:
@nis267 I indeed have multiple files because I have multiple cards. 
Depending on my setup I can have from one to three "cards" : 
- one intel card integrated to my laptop (10-intel.conf), 
- one external nvidia card (dockstation, 20-nvidia.conf) 
- the display link card (30-displaylink.conf)

Unfortunately changing this setup is not possible as I use it for work.

An I did try all combination possible : one file, two file (every combination), all at the same time and nothing worked properly for the external display link screen. 
The closest to a working system that I have ATM is the last setup I sent which gives me a black screen that looks to be holding something as I can move windows to it but the screen remains black.

But this same setup was working fine before upgrading to Debian Buster so there's no reason it shouldn't work...

Erwin-D, 2020-05-04:
Try this solution: https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only
This has solved similar problems in the past.

rrajkomar, 2020-05-07:
I will give it a shot asap.
Unfortunately with the current state of things, I might not be at my office for a while.

closed 2020-07-19 by Barabazs