AdnanHodzic/displaylink-debian #302: 2 monitors: 5120x1440 plus 3840x2160 resolution / or better triple monitor support

zenaan, 2019-09-06:
DisplayLink does not support 5120x1440 resolution monitors:

 https://support.displaylink.com/forums/287786-displaylink-feature-suggestions/suggestions/38503027-add-support-for-5120x1440-resolution

 https://www.dell.com/support/article/au/en/aubsd1/sln314416/dell-u4919dw-system-requirements-and-supported-configuration?lang=en

And 8K resolution is now purchasable albeit if you have four 4K HDMI ports to connect to it:

 https://www.samsung.com/au/tvs/qled-q900r-row/QA65Q900RBWXXY/

Trying to configure:
 - 3x DELL D6000 USB3 (plain, not thunderbolt) docks (which have 2 DP outputs (up to 4K 60Hz) and 1 HDMI output (up to 4K 30Hz)),
 - to attach to aging Lenovo x220 laptops (single USB3 port, inbuilt DP)
 - on Debian 10 Buster stable
 - to a 4K (LG) monitor, and a 5120x1440 (Samsung) monitor

Xorg in Debian 10 supports a maximum of 8192x8192 resolution Screen, but Xorg in the future should support larger setups:

 https://gitlab.freedesktop.org/xorg/xserver/issues/45

 https://gitlab.freedesktop.org/xorg/xserver/merge_requests/24

 https://patchwork.freedesktop.org/series/49663/

Placing the two monitors side by side, means:
5120+3840 = 8960, which is too wide for the maximum Xorg Screen.

Alternatively, placing one monitor virtually above the other, means:
2160+1440 = 3600, which should fit inside the maximum Xorg Screen possibility; this would be acceptable.

But since DisplayLink does not support 5120x1440, the only option in this setup is to use PBP/PIP picture by picture support on the 5120 monitor, giving it two 2560x1440 inputs side by side (quite sub optimal for refresh tearing, but adequate for office work).

So this means a 4K monitor either above (or below) two 2560x1440 monitors, which means an Xorg Screen of 5120x3600.

Xorg no longer has an /etc/X11/xorg.conf configured by default, auto configuring everything.

Xorg auto configures a "Screen" based on the monitors it detects, which in the above setup, is a 2560 monitor and 4K monitor, and so by the time X has started and I'm running the "xrandr --setprovideroutputsource 1 0" and "xrandr --setprovideroutputsource 2 0" commands, X does not allow a full 5120x3600 for some reason.

I am not able to configure an xorg.conf - I got a template out of X :7 -configure and tried to modify that, but many attempts so far have failed.

Ideally X would have an option here to specify the Screen section, -without- having to do all the other sections to tie the Screen section to.

zenaan, 2019-09-08:
See also:

xrandr fails to change virtual-screen size when required
https://gitlab.freedesktop.org/xorg/app/xrandr/issues/17

zenaan, 2019-09-08:
See also here for hints at creating xorg.conf snippets for extreme multi head setup:

ArchWiki: Extreme Multihead
https://wiki.archlinux.org/index.php/Extreme_Multihead

Barabazs, 2020-07-18:
This is clearly not an issue with this installer.

closed 2020-07-18 by Barabazs