AdnanHodzic/displaylink-debian #303: Script Freeze

Exomancer, 2019-09-10:
I had emailed you on Sep 5, 2019 and you told me to submit this as a bug on Sep 9, 2019.  Thank you for getting back to me.  

The problem is that I ran the script and it froze for several hours at:

```
Building EVDI kernel module with DKMS
[Cursor]
```

I ran the script again using the reinstall option and it performed a system shutdown.

I restarted the system and nothing was wrong so I waited for your reply.  I assume I should start over again.

The debug is from today, Sep 9, 2019.  The computer has been restarted several times since the script freeze.  The debug output is below:
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 4.15.0-60-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"                                                                                                                   
EndSection                                                                                                                                
                                                                                                                                          
-------------------- Monitors --------------------                                                                                        
                                                                                                                                          
Providers: number : 1                                                                                                                     
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel                         
                                                                                                                                          
-------------------------------------------------------------------                                                                       
                                                                                                                                          
Use this information when submitting an issue (http://bit.ly/2GLDlpY)                                                                     
                                                                                                                                          
-------------------------------------------------------------------

```

fragov, 2019-09-17:
I have same problem
``
at: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Ubuntu
Release: disco
Kernel: 5.0.0-27-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): NVIDIA Corporation GM108M [GeForce 940MX] (rev a2)
X11 version: 1.20.4-1ubuntu3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x66 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x3f cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:modesetting

fragov, 2019-09-17:
Problem is caused by UEFI Security boot. When I tried to build evdi myself, I've got a window in terminal, tell me I have to enter password for UEFI. I believe this freezes install.

fragov, 2019-09-17:
Right after I disabled UEFI Security script freezes no more.

closed 2020-07-18 by Barabazs