AdnanHodzic/displaylink-debian #304: display issue resolved - parrot security os

Rajamohan324, 2019-09-10:
for issue is resloved and display is coming .

┌─[root@parrot]─[~/displaylink-debian]
└──╼ #sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 5.2.0-2parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

janlueders, 2019-10-18:
Same here but with and Intel 620

during the insall progress it is possible to see whats may the Error.
First the linux-headers for parrot would be found and after the downloading they are missed

// INSTALL PRINT OUT
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.2.0-2parrot1-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------
Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------

Installing driver version: 5.2

DisplayLink Linux Software 5.2.14 install script called: install
Distribution discovered: Parrot GNU/Linux 4.7
Unsatisfied dependencies. Missing component: Linux headers for running kernel, 5.2.0-2parrot1-amd64.
This is a fatal error, cannot install DisplayLink Linux Software.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

// DEBUG PRINT OUT
--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 5.2.0-2parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

janlueders, 2019-10-18:
> for issue is resloved and display is coming .
>
> ┌─[root@parrot]─[~/displaylink-debian]
> └──╼ #sudo ./displaylink-debian.sh --debug
>
> Starting Debug ...
>
> Did you read Post Installation Guide? [...]

How you solve it?

closed 2020-07-18 by Barabazs