AdnanHodzic/displaylink-debian #311: No screen found after driver installation/System freezes when activating second screen

nikur, 2019-10-06:
Hi Adnan,

I bought a USB 3.0 docking station for my laptop and installed the latest DisplayLink drivers (version 5.2) I downloaded from the DisplayLink website. However, trying to activate the external screen always caused my systems to freeze, even Ctrl-Alt-Backspace didn't work anymore. I looked up a few solutions on the web but apparently none of them worked. I also tried disabling the "PageFlip" option for the modesetting driver but this caused my system to not even boot to the desktop when the external screen was connected.

Then I uninstalled the drivers using the standard uninstaller and deleted all addtional config files I had created trying to solve this issue. I then cloned you repo and ran the install script. However, now the external screen is not even detected by xrandr. I have an "Intel HD Graphics 5500 (rev 09)" installed and seen your solution on the troubleshooting page but I'm using X.org 1.18 rather than 1.20 (which the solution pertains to).

Could you take a look at my issue and tell me if this problem can be solved at all? Do I have to update to Linux Mint 18.3 in order to get my docking station working?

This is the debug output of the installer script:
`--------------- Linux system info ----------------

Distro: LinuxMint
Release: serena
Kernel: 4.4.0-165-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Integrated
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.18.4-0ubuntu0.8
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel

-------------------------------------------------------------------`

Many thanks in advance!

Nikur

Naegolus, 2020-01-04:
Hi Guys,

thanks for the amazing script!

I tried to update the whole "DL system" by uninstalling and reinstalling it.
Unfortunately my system also freezes at the moment.

```
--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-6-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 5917 (rev 07)
VGA (3D): 
X11 version: 1.19.2-1+deb9u4
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 4
Provider 0: id: 0x81 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting
Provider 1: id: 0x19b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x177 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

I can execute the following commands:
```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
```
Then I see the displays in debians display settings, but as soon as i execute this:
```
xrandr --output DVI-I-3-3 --mode 1920x1080 --pos 0x0 --rotate normal --output DVI-I-1-1 --mode 1920x1080 --pos 3840x0 --rotate normal --output eDP-1 --mode 1920x1080 --pos 3840x1080 --rotate normal --output DVI-I-2-2 --primary --mode 1920x1080 --pos 1920x0 --rotate normal --output DP-2 --off --output DP-1 --off
```
Or try to activate a display in debians display settings, the system freezes.

Naegolus, 2020-02-04:
Hi. I solved my Issue by upgrading my system to Debian Buster.

But then I had another Issue. The providers were absent.
This was because wayland was active. I switched to X11 and now it finally works.

Thank you Adnan for your script.

One suggestion: Stop the script with an error message if simple prerequisites are not met.
if [ ! "$(echo $XDG_SESSION_TYPE)" = "x11" ]; then ...

Or add it in the debug information. Otherwise people like me will post senseless Issues here ^^

Barabazs, 2020-07-18:
Thank you for the update. You're always welcome to provide a fix and send a PR.

closed 2020-07-18 by Barabazs