AdnanHodzic/displaylink-debian #318: Installation freezes at "Building EVDI kernel module with DKMS" on Ubuntu 19.10

michaeldfallen, 2019-10-31:
First installation freezes at the step `Building EVDI kernel module with DKMS`. I can kill the terminal and try again, which then skips past the building step and installs but when a monitor is connected it will show with heavy distortion and be entirely unusable.

I can't seem to get any logs from that step to debug why it's freezing.

```
sudo ./displaylink-debian.sh --install

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.3.0-19-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1369
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2019-10-31 10:51:46--  https://www.displaylink.com/downloads/file?id=1369
Resolving www.displaylink.com (www.displaylink.com)... 54.77.143.153, 34.252.102.94, 52.214.95.115, ...
Connecting to www.displaylink.com (www.displaylink.com)|54.77.143.153|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1572520295&Signature=UMAw0WbkZ2AJovCg%2Fr7Z6RukJx8%3D [following]
--2019-10-31 10:51:46--  http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1572520295&Signature=UMAw0WbkZ2AJovCg%2Fr7Z6RukJx8%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.106.27
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.106.27|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12768730 (12M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.2.zip’

DisplayLink_Ubuntu_5.2.zip            100%[=========================================================================>]  12.18M  14.3MB/s    in 0.9s    

2019-10-31 10:51:47 (14.3 MB/s) - ‘DisplayLink_Ubuntu_5.2.zip’ saved [12768730/12768730]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------

Installing driver version: 5.2

DisplayLink Linux Software 5.2.14 install script called: install
Distribution discovered: Ubuntu 19.10
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS

```

broilogabriel, 2019-11-01:
I'm having the same issue on Mint:

```
-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------

Installing driver version: 5.2

DisplayLink Linux Software 5.2.14 install script called: install
Distribution discovered: Linux Mint 19.2 Tina
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
```

Previously when I tried to install the standard displaylink driver from the official website I got a freeze like that too. Reinstalling `DKMS` and retrying worked but the driver didn't help with my issue of a Lenovo dock not working in Linux, that's why I'm trying this script since it's supposed to help other Linux distros.

solaris765, 2019-11-14:
Don't know if this is exactly the solution for you guys. But for me (kubuntu 19.10) I had upgraded my bios which re-enabled secure boot without my knowledge. 

I noticed when I ran `apt-get dist-upgrade` and my system prompted me to set a secure boot key as per the instruction on [displaylink's support site](https://support.displaylink.com/knowledgebase/articles/683672-my-displaylink-device-does-not-work-on-ubuntu)

I ended up testing with both secure boot disabled and with the key I configured and it all seems to work now. 

In regards to the DKMS build failing, I had no issues when secure boot was disabled, but on my current install I ended up running this script until it hung on the build then ran the displaylink script in another terminal from their site before cancelling this one and rebooting.

advance512, 2019-12-08:
> Reinstalling `DKMS` and retrying worked but the driver didn't help with my issue of a Lenovo dock not working in Linux, that's why I'm trying this script since it's supposed to help other Linux distros.

@broilogabriel - did you get this working on Ubuntu 19.10? If so, what did you do?
I'm using a Lenovo Thunderbolt 3 Dock and can't get the external displays to work. I am wondering whether the issue is DisplayLink, or Wayland as others claims.
https://forums.lenovo.com/t5/Ubuntu/P1-Thunderbolt-3-dock-Ubuntu-Linux-18-10-2-external-displays/m-p/4415426#M22

Care to share your conclusions with us?

advance512, 2019-12-08:
@broilogabriel I had a small breakthrough..

I have a Lenovo P1 Gen2 (with a Quadro T2000/PCIe/SSE2) with a freshly installed Ubuntu 19.10.
I also have a Lenovo Thunderbolt 3 Dock, with 2 monitors connected to it - one via DisplayPort and one via HDMI.

1. Using `prime-select intel` does not power off the nVidia GPU. This means that the power draw is very high, and my batter lasts about 30-50% of what it could last without the GPU. I don't always need the GPU, after all. The following hack ([url]https://github.com/stockmind/dell-xps-9560-ubuntu-respin/issues/8#issuecomment-389292575[/url]) is necessary to get the GPU to actually turn off.

2. However, using `prime-select intel` (or even `prime-select on-demand`) causes the external monitors to not be detected at all. It took me a while to figure out this was the culprit - I thought it might've been something related to DisplayLink (apparently not necessary at all), to X11 vs Wayland, to the nVidia drivers, Secure Boot, an X11 configuration of some sort, gdm3 vs lightdm, etc. But no, the only difference required to get the external monitors to work is to set `prime-select nvidia` and reboot.

Ideally, I'd like to have `prime-select on-demand` powering off the nVidia GPU until it is needed, and the exterior monitors working consistently.

For now, my solution is to uninstall DisplayLink completely, use basically stock Ubuntu 19.10 with `prime-select nvidia` selected (and no hack in use), and make sure that the "Extend boot by 5 seconds" option is selected in the BIOS. This means the battery last much less, but I guess this is fine for now.

Barabazs, 2020-07-18:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-07-18 by Barabazs

martinrebane, 2023-02-05:
I had the same issue but figured out that the "freeze" on evdi installation was not a freeze, but display issue. It intends to poll you about firmware signing password but terminal graphics is buggy (perhaps works at certain resolution only?) so the interactive terminal prompt does not appear.

To fix this: first uninstall all you installed and reboot (otherwise it wont start proper module installation again). For example if you have an official driver:
```
sudo displaylink-installer uninstall
sudo apt remove dkms
```
And as you are working with kernel here, then reboot!

Now install dkms again `sudo apt install dkms` and install either official driver or a community one as per instructions. Then you get to the part where it appearantly freezes on `Building EVDI kernel module with DKMS`. If nothing happens, wait 1 minute and put the terminal window on full screen or smaller size depending on what it is at the moment (if small, put to full screen and vice versa). Redrawing the screen should display a new prompt. The first one just asks you to press enter upon some message. Press ENTER and resize the terminal window again - then you see the next prompt. If you do not see anything, press ENTER, resize. Repeat, until you see something. When it asks you for a secure boot password, enter it (it must be 8 chars long, it is a new one time password, not your user password). Due to display issues, the **cursor is not at the password line, you do not see any interaction**, but it works anyway. Press ENTER after 8 chars. Resize again. Re-enter password the same way. Resize. When done, it exits the interactive mode and goes back to the terminal.

The installation finishes. Reboot and the computer prompts you to Continue boot or Enroll MOK key, select Enroll MOK and you can re-enter your one-time password to finish installation. And done!

It just an annoying display issue, not a real freeze. And there is actually a warning about it when the installation script starts. I think they should fix it :)

25Ericcheong, 2023-02-25:
> I had the same issue but figured out that the "freeze" on evdi installation was not a freeze, but display issue. It intends to poll you about firmware signing password but terminal graphics is buggy [...]

This works and I appreciate you for doing this cause it really helped me when I was trying to connect my AOC monitor to my linux system (my first time ever doing this setup)

juliusz-cwiakalski, 2023-03-12:
thx @martinrebane - was struggling with exactly same issue and your workaround resolved it :1st_place_medal: