AdnanHodzic/displaylink-debian #320: Driver stops working several times a day

tgaertner, 2019-11-04:
Randomly my two external screen (connected via the DisplayLink) disappear from the system. However they are still showing the last displayed state and have connection to the DisplayLink. 
Other connected stuff like mouse, keyboard and external Harddrive + Cam still works (that stuff does not require the driver).
To get my external monitors back I can restart the DisplayLink Device and all works again, but this is pain, because of reorganizing windows and remounting external HardDrive.

Debug:

--------------- Linux system info ----------------

Distro: LinuxMint
Release: tina
Kernel: 4.15.0-66-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1558:1325]
VGA: Intel Corporation Device 3ea0 (rev 02)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x9f cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 2 name:modesetting
Provider 1: id: 0x78 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

tgaertner, 2019-11-05:
Also connecting the DisplayLink after Login logs me out from Session. When DesplayPort is already connecten I have to login twice. Don't know if this is related...

dalamanster, 2019-11-05:
I have the same problem... Also have to login twice. I have to rearrange three screens after this restart (i am using "arandr" tool - it makes it easier)

tgaertner, 2019-11-05:
It's the same issue with the original driver from i-tec - that's why I moved to this one... But because I use LM18 and not "Ubuntu" the support adviced me to use this driver.

closed 2020-07-18 by Barabazs