AdnanHodzic/displaylink-debian #322: Monitors not detected after installing drivers

dietermijle, 2019-11-07:
The monitors are not detected after installation.

The DisplayLink info shows no information for both the service status and the EVDI service version.
Not sure if this relates to my issue.

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Ubuntu
Release: eoan
Kernel: 5.3.0-19-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 530 (rev 06)
VGA (3D): 
X11 version: 1.20.5+git20191008-0ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4b cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 8 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

Non the less when executing _lsusb_ command, the DisplayLink is included in the list

`Bus 004 Device 005: ID 17e9:600a DisplayLink`

Any idea what can cause this issue?

Gustry, 2019-11-23:
Same for me, since I installed this driver, external screens (HDMI or VGA) on my USB-C adaptor won't connect anymore.

jcdan3, 2019-12-22:
Have you tried to modify the file etc/X11/xorg.conf.d/20-displaylink.conf,
as explained in the post installation guide?

closed 2020-07-18 by Barabazs