AdnanHodzic/displaylink-debian #326: blank screen Startech USB32HDDVII works under win10 not under Debian9 NB laptop LCD+HDMI monitor off laptops hdmi aleady

kalpha2, 2019-11-28:
-------------------------------------------------------------------
Starting Debug ...
Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
--------------- Linux system info ----------------
Distro: Debian
Release: stretch
Kernel: 4.9.0-11-amd64
---------------- DisplayLink info ----------------
Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1
------------------ Graphics card -----------------
Vendor: i915
Subsystem: Ltd/Shuttle
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection 

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection
-------------------- Monitors --------------------
Providers: number : 2
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:Intel
Provider 1: id: 0x150 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
-------------------------------------------------------------------
Use this information when submitting an issue (http://bit.ly/2GLDlpY)
-------------------------------------------------------------------

lspci
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)

lsusb
Bus 001 Device 005: ID 17e9:4301 DisplayLink 

I've tried the two options in the 20-displaylink.conf file and also copied to /usr/share/X11/xorg.conf.d/ as well.

below is accepted no messages
$ xrandr --setprovideroutputsource 1 0

next 
$ xrandr --setprovideroutputsource 2 0
Could not find provider with index 2

also tried 
$ xrandr --setprovideroutputsource 0 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x47
  Serial number of failed request:  16
  Current serial number in output stream:  17

Barabazs, 2020-07-18:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

kalpha2, 2020-07-19:
still an issue, please re-open

Barabazs, 2020-07-19:
Can you post a new debug log please.

kalpha2, 2020-07-20:
grabbed new version, got 
`$ sudo ./displaylink-debian.sh --debug 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: stretch
Kernel: 4.9.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd/Shuttle
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.19.2-1+deb9u5
X11 configs: /etc/X11/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
  Identifier "DisplayLink"
  Driver "modesetting"
  Option "PageFlip" "false"
EndSection 

#Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------`

I also tried as per some of the links tweaking 
/etc/X11/xorg.conf.d/20-displaylink.conf
with various combinations which most resulted in X not even starting.
and also removing completely, which allows X to start but no sign of monitor via displaylink working (and yes monitor works from other direct connections to units)

checking logs
`before 
user12@unit12:~$ sudo dmesg | grep Display
[    2.454406] usb 2-1.2: Manufacturer: DisplayLink
[   15.186933] evdi: [I] Task 707 (DesktopManagerE) of process 626 (DisplayLinkMana)
[  327.476371] evdi: [I] Task 707 (DesktopManagerE) of process 626 (DisplayLinkMana)

after
user12@unit12:~$ sudo dmesg | grep Display
[    2.454406] usb 2-1.2: Manufacturer: DisplayLink
[   15.186933] evdi: [I] Task 707 (DesktopManagerE) of process 626 (DisplayLinkMana)
[  327.476371] evdi: [I] Task 707 (DesktopManagerE) of process 626 (DisplayLinkMana)
[  382.696093] usb 2-1.2: Manufacturer: DisplayLink
[  383.865599] evdi: [I] Task 1646 (DesktopManagerE) of process 1639 (DisplayLinkMana)`

the /var/log/displaylink/ log files appear to be binary like, no readable text as such

these files also appeared at same time of boot.
PHL 288P6L_PHL08F2-2016.4.0000182D.edid
and
.dl.xml
which contains
`<?xml version="1.0" encoding="ISO-8859-1"?>
<DL_HIVE>
    <node name="HKEY_CURRENT_USER"/>
    <node name="HKEY_LOCAL_MACHINE">
        <node name="Software">
            <node name="DisplayLink">
                <node name="">
                    <node name="NivoSettings">
                        <node name="Edid">
                            <value name="USB_4301-YUAN412190108081590480^1">00FFFFFFFFFFFF00410CF2082D180000041A0103803E22782A5525A1574FA0280F5054BD4B00D1C081808140950F9500B30081C001014DD000A0F0703E80302035006D552100001AA36600A0F0701F80302035006D552100001A000000FC0050484C2032383850364C0A2020000000FD0017501E633C000A20202020202001DF020333F14C9004031F1301125D5E5F606123090707830100006D030C001000397820006001020367D85DC401788003E30F000C011D007251D01E206E2855006D552100001E8C0AD08A20E02D10103E96006D55210000184D6C80A070703E8030203A006D552100001A04740030F2705A80B0588A006D552100001A00000000A1</value>
                        </node>
                    </node>
                </node>
            </node>
        </node>
    </node>
</DL_HIVE>`

Barabazs, 2020-07-20:
I've edited your post for clarity.

Did you first uninstall and tried to reinstall with the latest installer?
Sorry if I wasn't clear...

kalpha2, 2020-07-20:
<<Did you first uninstall >>
No, how do I do that?
I just assumed run the install again would overwrite ...

Barabazs, 2020-07-20:
I'm not sure it does overwrite anything when you select install. Your installed versions are identical to those in 2019.

 1. `sudo ./displaylink-debian.sh > uninstall.log` 
And select the uninstall option, follow all steps.
2.  `sudo ./displaylink-debian.sh > install.log` 
And select the install option, follow all steps.
3. Test if it works
4. If it doesn't work report back with the content of `uninstall.log`, `install.log` and a new debug output

AdnanHodzic, 2020-07-20:
Just a suggestion, you can also use install/uninstall/debug functionality as parameters, i.e:

```
sudo ./displaylink-debian.sh --install
sudo ./displaylink-debian.sh --uninstall
sudo ./displaylink-debian.sh --debug
```
So one liners would also be an option, i.e:

```
sudo ./displaylink-debian.sh --install > install.log
```

kalpha2, 2020-07-20:
uninstall log
```
Checking dependencies

unzip is installed
linux-headers-4.9.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

DisplayLink Linux Software 5.2.14 install script called: uninstall
Distribution discovered: Debian GNU/Linux 9.13 (stretch)
Uninstalling
Removing EVDI from kernel tree, DKMS, and removing sources.
Stopping displaylink-driver systemd service
Removing suspend-resume hooks
Removing udev rule
Removing Core folder

Uninstallation steps complete.
Please note that the evdi kernel module is still in the memory.
A reboot is required to fully complete the uninstallation process.
Removing Displaylink Xorg config file
```

install log 
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.9.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: 
Downloading DisplayLink Ubuntu driver:

[... 82 lines cut ...]
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:49:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .fault = evdi_gem_fault,
           ^~~~~~~~~~~~~~
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:49:11: note: (near initialization for ‘evdi_gem_vm_ops.fault’)
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:86:12: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .unload = evdi_driver_unload,
            ^~~~~~~~~~~~~~~~~~
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:86:12: note: (near initialization for ‘driver.unload’)
/home/user12/displaylink-debian/evdi/module/evdi_drv.c: In function ‘evdi_platform_probe’:
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:191:2: error: implicit declaration of function ‘drm_dev_put’ [-Werror=implicit-function-declaration]
  drm_dev_put(dev);
  ^~~~~~~~~~~
/home/user12/displaylink-debian/evdi/module/evdi_drv.c: In function ‘evdi_platform_remove’:
/home/user12/displaylink-debian/evdi/module/evdi_drv.c:201:2: error: implicit declaration of function ‘drm_dev_unplug’ [-Werror=implicit-function-declaration]
  drm_dev_unplug(drm_dev);
  ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[5]: *** [/home/user12/displaylink-debian/evdi/module/evdi_drv.o] Error 1
make[4]: *** [_module_/home/user12/displaylink-debian/evdi/module] Error 2
make[3]: *** [sub-make] Error 2
make[2]: *** [all] Error 2
make[1]: *** [module] Error 2
make: *** [all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory
```

Barabazs, 2020-07-20:
What does `lsmod | grep evdi` and `modinfo evdi` output?
Is secureboot off?

kalpha2, 2020-07-20:
sudo lsmod | grep evdi
returns nothing

sudo modinfo evdi
modinfo: ERROR: Module evdi not found.

Secure Boot
Disabled

Barabazs, 2020-07-21:
Looks like evdi dropped support for kernels < 4.15 
https://github.com/DisplayLink/evdi/commit/d242444c656b5ba015267895e1ecc70fd1112371

closed 2020-08-05 by Barabazs