AdnanHodzic/displaylink-debian #331: a start job is running for udev wait for complete device initialization. Ubuntu Pi ARM

kingspeech64, 2019-12-10:
After installing the driver **boot** hangs 2 minutes on the message:
a start job is running for udev wait for complete device initialization (Xmin Ys/ 2min 59s)
followed by 'FAILED'

**Debug is:**
-------------------------------------------------------------------

Starting Debug ...

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: eoan
Kernel: 5.3.0-1014-raspi2

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: 
Subsystem: 
VGA: 
VGA (3D): 
X11 version: 1.20.5+git20191008-0ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x41 cap: 0x0 crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

**sudo systemctl status /etc/udev/rules.d/displaylink-driver.service** 
● displaylink-driver.service - DisplayLink Driver Service
   Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2019-12-10 14:58:49 UTC; 4s ago
  Process: 17294 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install evdi/5.2.14 && modprobe evdi) (code=exited, status=0/SUCCESS)
  Process: 17297 ExecStart=/opt/displaylink/DisplayLinkManager (code=exited, status=203/EXEC)
 Main PID: 17297 (code=exited, status=203/EXEC)

Dec 10 14:58:54 ubuntu systemd[1]: displaylink-driver.service: Service RestartSec=5s expired, scheduling restart.
Dec 10 14:58:54 ubuntu systemd[1]: displaylink-driver.service: Scheduled restart job, restart counter is at 3137.
Dec 10 14:58:54 ubuntu systemd[1]: Stopped DisplayLink Driver Service.
Dec 10 14:58:54 ubuntu systemd[1]: Starting DisplayLink Driver Service...
Dec 10 14:58:54 ubuntu systemd[1]: Started DisplayLink Driver Service.
Dec 10 14:58:54 ubuntu systemd[17310]: displaylink-driver.service: Failed to execute command: Exec format error
Dec 10 14:58:54 ubuntu systemd[17310]: displaylink-driver.service: Failed at step EXEC spawning /opt/displaylink/DisplayLinkManager: Exec format error
Dec 10 14:58:54 ubuntu systemd[1]: displaylink-driver.service: Main process exited, code=exited, status=203/EXEC
Dec 10 14:58:54 ubuntu systemd[1]: displaylink-driver.service: Failed with result 'exit-code'.

**dash /opt/displaylink/DisplayLinkManager** 
/opt/displaylink/DisplayLinkManager: 6: Syntax error: word unexpected (expecting ")")

**file DisplayLinkManager**
DisplayLinkManager: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=5720db8b7a6202d64b2e90a92716f0d0292a74ca, stripped

**uname -a**
Linux ubuntu 5.3.0-1014-raspi2 #16-Ubuntu SMP Tue Nov 26 11:18:23 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux

I conclude that this is an architecture error. The driver should be support ARM. Is there another way as cross compiling DisplayLinkManager to aarch64? It isn't a permission problem. Also no bash/dash problem.

NOTE: I'm working with Raspberry Pi 4b with Ubuntu 19.10

shriniwas26, 2020-01-15:
I am facing the same issue. If somebody can help me understand the build process, I can attempt to cross compile.

emre-yildirim, 2020-02-03:
I also have the same issue.
I have toshiba ssd.
After downloading ubuntu from ubuntu web site, I've created a bootable usb disk within ubuntu and installed the experimental mode. using the uefi option of the usb boot.

and every time i restart my computer, i have to wait for 1.5 mins with the same error message

SnejPro, 2020-04-09:
Same issue on an Ubuntu 19.04 persistence Live Stick on amd64

Barabazs, 2020-07-18:
This doesn't look like an issue related to the installer.

closed 2020-07-18 by Barabazs