AdnanHodzic/displaylink-debian #333: system freezes when connecting display

jonathanschiefer, 2019-12-13:
Hi,

I have a problem when using a USB-C dock to connect an external display to my notebook: when I'm connecting the display, the external monitor shows the screen but the system freezes, no reaction, no switching to tty2.

The ouput of the debug mode is (when display is not connect, not possible when display is connected):
Can't open display 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.3.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 4.4.24
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.6-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

-------------------------------------------------------------------

Someone any idea?
Thanks in advance!

callmebob2016, 2019-12-15:
I am experiencing the same issue since I upgraded from ubuntu 19.04 to 19.10, kernel 5.3.0-24 currently. I never had it on 19.04. I was, however, using the 5.0.5 kernel with 19.04. Perhaps downgrading kernel would help? 

One thing I noticed (although I need to test it more, so don't quote me on this), is that it seems to work fine when I start my laptop connected to power supply. Previously most of the time it was working from battery and I would have to give it a hard reset 1-3 times before it booted and didn't freeze. But again, this needs more data points.

jonathanschiefer, 2019-12-15:
You're right, when I boot the laptop with the dock connected (which has power delivery) the external display works. However if I make a change to the settings, the system still freezes...

callmebob2016, 2019-12-16:
I'm lucky then. In my case once my screens start with `no freeze` everything works correctly no matter what I change.

afdaniels, 2019-12-17:
Was this resolved?

callmebob2016, 2019-12-17:
> You're right, when I boot the laptop with the dock connected (which has power delivery) the external display works. However if I make a change to the settings, the system still freezes...

Unfortunately after more testing I am sad to report power supply doesn't make a difference. I must have had a lucky run.

jonathanschiefer, 2019-12-21:
The Problem has been solved using a DisplayPort cable instead of HDMI...

closed 2019-12-21 by jonathanschiefer

callmebob2016, 2020-01-26:
> The Problem has been solved using a DisplayPort cable instead of HDMI...

I had been using DisplayPort and HDMI combination the whole time and still had the problem.

After few weeks of observation and testing I think I know what is happening in my case. It is to do with the amount of USB devices plugged into the docking station. In particular power-heavy devices, like my Bose 2.1 speaker system. If I minimize/eliminate additional USB devices plugged into the docking station I seem to have no issues whatsoever. 

Hope this helps someone.