AdnanHodzic/displaylink-debian #346: Sorry, not for: Linux PQMC-13 5.4.10-050410-generic #202001091038 SMP Thu Jan 9 10:41:11 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

BluemchenBlau, 2020-01-09:
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: Datei oder Verzeichnis nicht gefunden
--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.10-050410-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

###### Reinstall:
Select a key: [i/d/r/u/q]: r

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.10-050410-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: Datei oder Verzeichnis nicht gefunden
Removing Displaylink Xorg config file

-------------------------------------------------------------------

Performing clean-up

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.10-050410-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1369
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-01-09 15:13:09--  https://www.displaylink.com/downloads/file?id=1369
Auflösen des Hostnamens www.displaylink.com (www.displaylink.com) … 34.252.102.94, 18.203.109.3, 54.77.143.153, ...
Verbindungsaufbau zu www.displaylink.com (www.displaylink.com)|34.252.102.94|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 303 See Other
Platz: http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1578580380&Signature=muH%2B9TdbY3sJatizCo73S78rzG4%3D [folgend]
--2020-01-09 15:13:09--  http://assets.displaylink.com/live/downloads/software/f1369_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.2.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1578580380&Signature=muH%2B9TdbY3sJatizCo73S78rzG4%3D
Auflösen des Hostnamens assets.displaylink.com (assets.displaylink.com) … 52.218.106.147
Verbindungsaufbau zu assets.displaylink.com (assets.displaylink.com)|52.218.106.147|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 12768730 (12M) [application/octet-stream]
Wird in »DisplayLink_Ubuntu_5.2.zip« gespeichert.

DisplayLink_Ubuntu_ 100%[===================>]  12,18M  4,94MB/s    in 2,5s    

2020-01-09 15:13:12 (4,94 MB/s) - »DisplayLink_Ubuntu_5.2.zip« gespeichert [12768730/12768730]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------

Installing driver version: 5.2

DisplayLink Linux Software 5.2.14 install script called: install
Distribution discovered: Ubuntu 18.04.3 LTS
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
ERROR (code 3): Failed to build evdi/5.2.14. Consult /var/lib/dkms/evdi/5.2.14/build/make.log for details..
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

sed: /lib/systemd/system/displaylink-driver.service kann nicht gelesen werden: Datei oder Verzeichnis nicht gefunden
Enable and start displaylink-driver service
Failed to enable unit: Unit file displaylink-driver.service does not exist.
Failed to start displaylink-driver.service: Unit displaylink-driver.service not found.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.2.zip" file
Removing redundant: "5.2" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes

harmjan, 2020-01-13:
I'm having the same problem after upgrading kernel on Debian.

Last time I got it to work I did have to change the xorg file from:
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
to
```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
```

# Debug
```
sudo ./displaylink-debian.sh 
[sudo] password for harmjan: 
lspci: Unable to load libkmod resources: error -12
lspci: Unable to load libkmod resources: error -12

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.4.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: 
EVDI service version: 

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1028:06dc]
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
[... 176 lines cut ...]
make -f /usr/src/linux-headers-5.4.0-2-common/Makefile syncconfig
if [ -f /usr/src/linux-headers-5.4.0-2-common/.config -o \
	 -d /usr/src/linux-headers-5.4.0-2-common/include/config -o \
	 -d /usr/src/linux-headers-5.4.0-2-common/arch/x86/include/generated ]; then \
	echo >&2 "***"; \
	echo >&2 "*** The source tree is not clean, please run 'make mrproper'"; \
	echo >&2 "*** in /usr/src/linux-headers-5.4.0-2-common";\
	echo >&2 "***"; \
	false; \
fi
make -f /usr/src/linux-headers-5.4.0-2-common/scripts/Makefile.build obj=scripts/basic
ln -fsn /usr/src/linux-headers-5.4.0-2-common source
sh /usr/src/linux-headers-5.4.0-2-common/scripts/mkmakefile /usr/src/linux-headers-5.4.0-2-common
sh: 0: Can't open /usr/src/linux-headers-5.4.0-2-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.4.0-2-common/Makefile:513: outputmakefile] Error 127
make[3]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-5.4.0-2-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-2-common/scripts/basic/Makefile: No such file or directory
make[4]: *** No rule to make target '/usr/src/linux-headers-5.4.0-2-common/scripts/basic/Makefile'.  Stop.
make[3]: *** [/usr/src/linux-headers-5.4.0-2-common/Makefile:499: scripts_basic] Error 2
make[2]: *** [/usr/src/linux-headers-5.4.0-2-common/Makefile:677: include/config/auto.conf.cmd] Error 2
make[1]: *** [/usr/src/linux-headers-5.4.0-2-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-2-amd64'
make: *** [Makefile:22: all] Error 2
```
I don't think I had the same evdi build problem before and suspect that that is the root cause.

rubin110, 2020-01-19:
I believe I'm hitting the same problem as described above.

Debian Sid
Linux lines 5.4.0-1-amd64 #1 SMP Debian 5.4.6-1 (2019-12-27) x86_64 GNU/Linux

1mckenna, 2020-02-09:
Try with PR #361

AdnanHodzic, 2020-02-09:
PR #361 has been merged with `master`. Please pull the latest changes and try doing the same operation.

harmjan, 2020-02-11:
It works for me after pulling in master again, thanks!

Crashdummyy, 2020-02-20:
I get this error as well:

ERROR (code 3): Failed to build evdi/5.2.14. Consult /var/lib/dkms/evdi/5.2.14/build/make.log 

any workaround for this?

Barabazs, 2020-07-19:
Closing this issue due to inactivity.

@Crashdummyy  Please open a new issue if your issue still persists.

closed 2020-07-19 by Barabazs