AdnanHodzic/displaylink-debian #347: please include this patch for debian bullseye

asterlab, 2020-01-13:
please include this patch for debian bullseye

`diff --git a/.travis.yml b/.travis.yml
index 0a9c52b..932d18f 100644
--- a/.travis.yml
+++ b/.travis.yml
@@ -26,6 +26,8 @@ env:
   - KVER=5.0
   - KVER=5.1
   - KVER=5.2
+  - KVER=5.3
+  - KVER=5.4
   - KVER=master
 
 matrix:
diff --git a/Makefile b/Makefile
index b19fe8b..66f32fe 100644
--- a/Makefile
+++ b/Makefile
@@ -8,6 +8,14 @@ all:
 	CFLAGS="$(FLAGS)" $(MAKE) -C module $(MFLAGS)
 	CFLAGS="-I../module $(FLAGS) $(CFLAGS)" $(MAKE) -C library $(MFLAGS)
 
+install:
+	$(MAKE) -C module install
+	$(MAKE) -C library install
+
+uninstall:
+	$(MAKE) -C module uninstall
+	$(MAKE) -C library uninstall
+
 clean:
 	$(MAKE) clean -C module $(MFLAGS)
 	$(MAKE) clean -C library $(MFLAGS)
diff --git a/library/Makefile b/library/Makefile
index 03e9999..7da8e3b 100644
--- a/library/Makefile
+++ b/library/Makefile
@@ -2,17 +2,37 @@
 # Copyright (c) 2015 - 2019 DisplayLink (UK) Ltd.
 #
 
+INSTALL ?= install
+LN ?= ln
+RM ?= rm
+
 DEPS = evdi_ioctl.h
 CFLAGS := -I../module -std=gnu99 -fPIC $(CFLAGS)
 
-default: libevdi.so
+LIBVER := 1.6.4
+LIBABI := 0
+
+PREFIX ?= /usr/local
+LIBDIR ?= $(PREFIX)/lib
+
+default: libevdi.so.$(LIBVER)
 
 clean:
-	rm -rf *.o libevdi.so
+	rm -rf *.o libevdi.so.$(LIBVER)
 
 %.o: %.c $(DEPS)
-	$(CC) -c -o $@ $< $(CFLAGS)
+	$(CC) $(CFLAGS) -o $@ -c $<
+
+libevdi.so.$(LIBVER): evdi_lib.o
+	$(CC) $^ -shared -Wl,-soname,libevdi.so.$(LIBABI) -o $@ -lc -lgcc $(LDFLAGS)
 
-libevdi.so: evdi_lib.o
-	$(CC) $^ -o $@ -lc -lgcc -shared -Wl,-soname,libevdi.so.0
+install:
+	$(INSTALL) -d $(DESTDIR)$(LIBDIR)
+	$(INSTALL) -m 755 libevdi.so.$(LIBVER) $(DESTDIR)$(LIBDIR)/libevdi.so.$(LIBVER)
+	$(LN) -sf libevdi.so.$(LIBVER) $(DESTDIR)$(LIBDIR)/libevdi.so.$(LIBABI)
+	$(LN) -sf libevdi.so.$(LIBABI) $(DESTDIR)$(LIBDIR)/libevdi.so
 
+uninstall:
+	$(RM) -f $(DESTDIR)$(LIBDIR)/libevdi.so.$(LIBVER)
+	$(RM) -f $(DESTDIR)$(LIBDIR)/libevdi.so.$(LIBABI)
+	$(RM) -f $(DESTDIR)$(LIBDIR)/libevdi.so
diff --git a/module/Makefile b/module/Makefile
index 8281112..7605c34 100644
--- a/module/Makefile
+++ b/module/Makefile
@@ -43,22 +43,39 @@ obj-$(CONFIG_DRM_EVDI) := evdi.o
 else
 
 # kbuild against specified or current kernel
+RM ?= rm
+
 ifeq ($(KVER),)
 	KVER := $(shell uname -r)
 endif
 
+ifneq ($(RUN_DEPMOD),)
+	DEPMOD := /sbin/depmod -a
+else
+	DEPMOD := true
+endif
+
 ifeq ($(KDIR),)
 	KDIR := /lib/modules/$(KVER)/build
 endif
 
+MOD_KERNEL_PATH := /kernel/drivers/gpu/drm/evdi
+
 default: module
 
 module:
 	$(MAKE) -C $(KDIR) M=$$PWD
 
+install:
+	$(MAKE) -C $(KDIR) M=$$PWD INSTALL_MOD_PATH=$(DESTDIR) INSTALL_MOD_DIR=$(MOD_KERNEL_PATH) modules_install
+	$(DEPMOD)
+
 clean:
-	rm -rf *.o *.ko .tmp* .*.*.cmd Module.symvers evdi.mod.c modules.order
+	$(RM) -rf *.o *.ko .tmp* .*.*.cmd Module.symvers evdi.mod.c modules.order
 
+uninstall:
+	$(RM) -rf $(DESTDIR)/lib/modules/$(KVER)/$(MOD_KERNEL_PATH)
+	$(DEPMOD)
 
 endif # ifneq ($(KERNELRELEASE),)
 
diff --git a/module/evdi_drv.c b/module/evdi_drv.c
index e59d0a1..8478ceb 100644
--- a/module/evdi_drv.c
+++ b/module/evdi_drv.c
@@ -81,8 +81,12 @@ static void evdi_disable_vblank(__always_unused struct drm_device *dev,
 }
 
 static struct drm_driver driver = {
-	.driver_features = DRIVER_MODESET | DRIVER_GEM | DRIVER_PRIME
-			 | DRIVER_ATOMIC,
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
+		.driver_features = DRIVER_MODESET | DRIVER_GEM | DRIVER_PRIME
+						| DRIVER_ATOMIC,
+#else
+		.driver_features = DRIVER_MODESET | DRIVER_GEM | DRIVER_ATOMIC,
+#endif
 #if KERNEL_VERSION(4, 12, 0) > LINUX_VERSION_CODE
 	.load = evdi_driver_load,
 #endif
diff --git a/module/evdi_drv.h b/module/evdi_drv.h
index 0d67197..13fbfce 100644
--- a/module/evdi_drv.h
+++ b/module/evdi_drv.h
@@ -20,16 +20,20 @@
 #include <drm/drm_crtc_helper.h>
 #include <drm/drm_rect.h>
 #include <drm/drm_gem.h>
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
 #include <linux/reservation.h>
+#else
+#include <linux/dma-resv.h>
+#endif
 #include "evdi_debug.h"
 
 #define DRIVER_NAME   "evdi"
 #define DRIVER_DESC   "Extensible Virtual Display Interface"
-#define DRIVER_DATE   "20180913"
+#define DRIVER_DATE   "20191204"
 
 #define DRIVER_MAJOR      1
 #define DRIVER_MINOR      6
-#define DRIVER_PATCHLEVEL 1
+#define DRIVER_PATCHLEVEL 4
 
 struct evdi_fbdev;
 struct evdi_painter;
@@ -51,8 +55,13 @@ struct evdi_gem_object {
 	struct page **pages;
 	void *vmapping;
 	struct sg_table *sg;
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
 	struct reservation_object *resv;
 	struct reservation_object _resv;
+#else
+	struct dma_resv *resv;
+	struct dma_resv _resv;
+#endif
 };
 
 #define to_evdi_bo(x) container_of(x, struct evdi_gem_object, base)
@@ -112,8 +121,12 @@ uint32_t evdi_gem_object_handle_lookup(struct drm_file *filp,
 
 struct drm_gem_object *evdi_gem_prime_import(struct drm_device *dev,
 					     struct dma_buf *dma_buf);
-struct dma_buf *evdi_gem_prime_export(struct drm_device *dev,
-				      struct drm_gem_object *obj, int flags);
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
+struct dma_buf *evdi_gem_prime_export(__maybe_unused struct drm_device *dev,
+				struct drm_gem_object *obj, int flags);
+#else
+struct dma_buf *evdi_gem_prime_export(struct drm_gem_object *obj, int flags);
+#endif
 
 int evdi_gem_vmap(struct evdi_gem_object *obj);
 void evdi_gem_vunmap(struct evdi_gem_object *obj);
diff --git a/module/evdi_gem.c b/module/evdi_gem.c
index 2895d6c..7487565 100644
--- a/module/evdi_gem.c
+++ b/module/evdi_gem.c
@@ -61,7 +61,11 @@ struct evdi_gem_object *evdi_gem_alloc_object(struct drm_device *dev,
 		return NULL;
 	}
 
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
 	reservation_object_init(&obj->_resv);
+#else
+	dma_resv_init(&obj->_resv);
+#endif
 	obj->resv = &obj->_resv;
 
 	return obj;
@@ -254,8 +258,11 @@ void evdi_gem_free_object(struct drm_gem_object *gem_obj)
 
 	if (gem_obj->dev->vma_offset_manager)
 		drm_gem_free_mmap_offset(gem_obj);
-
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
 	reservation_object_fini(&obj->_resv);
+#else
+	dma_resv_fini(&obj->_resv);
+#endif
 	obj->resv = NULL;
 }
 
@@ -569,8 +576,11 @@ struct drm_gem_object *evdi_gem_prime_import(struct drm_device *dev,
 	return ERR_PTR(ret);
 }
 
-struct dma_buf *evdi_gem_prime_export(__maybe_unused struct drm_device *dev,
-				      struct drm_gem_object *obj, int flags)
+struct dma_buf *evdi_gem_prime_export(
+#if KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
+				__maybe_unused struct drm_device *dev,
+#endif
+				struct drm_gem_object *obj, int flags)
 {
 	struct evdi_gem_object *evdi_obj = to_evdi_bo(obj);
 	struct dma_buf_export_info exp_info = {
@@ -581,8 +591,8 @@ struct dma_buf *evdi_gem_prime_export(__maybe_unused struct drm_device *dev,
 		.resv = evdi_obj->resv,
 		.priv = obj
 	};
-
-#if KERNEL_VERSION(4, 9, 0) <= LINUX_VERSION_CODE
+#if KERNEL_VERSION(4, 9, 0) < LINUX_VERSION_CODE && \
+	KERNEL_VERSION(5, 4, 0) > LINUX_VERSION_CODE
 	return drm_gem_dmabuf_export(dev, &exp_info);
 #else
 	return dma_buf_export(&exp_info);`

asterlab, 2020-01-13:
[evdi-all-in-one-fixes.zip](https://github.com/AdnanHodzic/displaylink-debian/files/4053442/evdi-all-in-one-fixes.zip)

asterlab, 2020-01-13:
also solve this:
/usr/src/linux-headers-5.4.0-2-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-2-common/scripts/basic/Makefile: No such file or directory

with the patch applied the module compiles fine but cannot make it work with dkms... so that the script doesn't finish correctly and I have no displaylink.service installed nor displaylinkmanager program

AdnanHodzic, 2020-01-13:
I'm not sure when I'll have time to add this change myself. Hence if anyone could please submit this as a PR I'd greatly appreciate it.

rudazhan, 2020-01-15:
I'm also on Debian bullseye and evdi currently does not build successfully. This patch seems to be for the DisplayLink/evdi repo, which I do not know how to build from.

asterlab, 2020-01-16:
builds fine with the patch provided now, please include the patch in the script

1mckenna, 2020-01-29:
> builds fine with the patch provided now, please include the patch in the script

what were the steps you took to get this working?

asterlab, 2020-01-29:
Since I had already the displaylinkmanager in 

/opt/displaylink# ls 
DisplayLinkManager DisplayLinkManager.log DisplayLinkManager.old.log DisplayLinkManagerDirLock libevdi.so libusb-1.0.so.0.1.0 

since I had already setup /etc/modprobe.d/evdi.conf for my 4 monitors configuration (two are via two usb adapters) 
options evdi initial_device_count=2 

since I had already made and enabled the /etc/systemd/system/displaylink.service 
[Unit] 
Description=displaylink service 
Before=gdm3.service 

[Service] 
ExecStart=/opt/displaylink/DisplayLinkManager & 
KillMode=process 
Restart=on-failure 
RestartPreventExitStatus=255 
Type=notify 

[Install] 
WantedBy=multi-user.target 
Alias=displaylink.service 

since I had already executed systemctl daemon-reload 

I downloaded the evdi git in /usr/lib/modules/5.5.0-rc5-amd64/kernel/drivers/, 
cd /usr/lib/modules/5.5.0-rc5-amd64/kernel/drivers/evdi 
applied the patch, 
make 
reboot 
depmod -a 
reboot 

Renato Gallo

rudazhan, 2020-02-01:
Made it with Debian bullseye 5.4.0-3-amd64. Thanks @asterlab !

loiselqw, 2020-02-02:
It seems I am having exactly the same issue..

- DisplayLink Linux Software 5.2.14 install script called: install
- Distribution discovered: Debian GNU/Linux bullseye/sid

`Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
ERROR (code 3): Failed to build evdi/5.2.14. Consult /var/lib/dkms/evdi/5.2.14/build/make.log for details..`

`DKMS make.log for evdi-5.2.14 for kernel 5.4.0-3-amd64 (x86_64)
So 2. Feb 15:56:24 CET 2020
make KBUILD_VERBOSE=1 SUBDIRS=/var/lib/dkms/evdi/5.2.14/build SRCROOT=/var/lib/dkms/evdi/5.2.14/build CONFIG_MODULE_SIG= -C /lib/modules/5.4.0-3-amd64/build modules
make[1]: Verzeichnis „/usr/src/linux-headers-5.4.0-3-amd64“ wird betreten
make -C /usr/src/linux-headers-5.4.0-3-amd64 -f /usr/src/linux-headers-5.4.0-3-common/Makefile modules
make -f /usr/src/linux-headers-5.4.0-3-common/Makefile syncconfig
if [ -f /usr/src/linux-headers-5.4.0-3-common/.config -o \
	 -d /usr/src/linux-headers-5.4.0-3-common/include/config -o \
	 -d /usr/src/linux-headers-5.4.0-3-common/arch/x86/include/generated ]; then \
	echo >&2 "***"; \
	echo >&2 "*** The source tree is not clean, please run 'make mrproper'"; \
	echo >&2 "*** in /usr/src/linux-headers-5.4.0-3-common";\
	echo >&2 "***"; \
	false; \
fi
make -f /usr/src/linux-headers-5.4.0-3-common/scripts/Makefile.build obj=scripts/basic
ln -fsn /usr/src/linux-headers-5.4.0-3-common source
sh /usr/src/linux-headers-5.4.0-3-common/scripts/mkmakefile /usr/src/linux-headers-5.4.0-3-common
sh: 0: Can't open /usr/src/linux-headers-5.4.0-3-common/scripts/mkmakefile
make[3]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:513: outputmakefile] Fehler 127
make[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
/usr/src/linux-headers-5.4.0-3-common/scripts/Makefile.build:42: /usr/src/linux-headers-5.4.0-3-common/scripts/basic/Makefile: Datei oder Verzeichnis nicht gefunden
make[4]: *** Keine Regel, um „/usr/src/linux-headers-5.4.0-3-common/scripts/basic/Makefile“ zu erstellen.  Schluss.
make[3]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:499: scripts_basic] Fehler 2
make[2]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:677: include/config/auto.conf.cmd] Fehler 2
make[1]: *** [/usr/src/linux-headers-5.4.0-3-common/Makefile:179: sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-5.4.0-3-amd64“ wird verlassen
make: *** [Makefile:22: all] Fehler 2
`

could you please describe how the patch needs to be used to make build process happen.

in which directory which file has to be changed which way.

Thx a lot in adv

asterlab, 2020-02-03:
@loiselqw 

As I said in this thread: 

Since I had already the displaylinkmanager in 

/opt/displaylink# ls 
DisplayLinkManager DisplayLinkManager.log DisplayLinkManager.old.log DisplayLinkManagerDirLock libevdi.so libusb-1.0.so.0.1.0 

since I had already setup /etc/modprobe.d/evdi.conf for my 4 monitors configuration (two are via two usb adapters) 
options evdi initial_device_count=2 

since I had already made and enabled the /etc/systemd/system/displaylink.service 
[Unit] 
Description=displaylink service 
Before=gdm3.service 

[Service] 
ExecStart=/opt/displaylink/DisplayLinkManager & 
KillMode=process 
Restart=on-failure 
RestartPreventExitStatus=255 
Type=notify 

[Install] 
WantedBy=multi-user.target 
Alias=displaylink.service 

since I had already executed systemctl daemon-reload 

I downloaded the evdi git in /usr/lib/modules/5.5.0-rc5-amd64/kernel/drivers/, 
cd /usr/lib/modules/5.5.0-rc5-amd64/kernel/drivers/evdi 
applied the patch, 
make 
reboot 
depmod -a 
reboot 

Renato Gallo

giggls, 2020-02-05:
Hm, so what would one do if /opt/displaylink is not even there yet?

asterlab, 2020-02-05:
We could make the install create it 

Renato Gallo

1mckenna, 2020-02-08:
I have created a PR for fixing this, you should be able to clone my repo and use it until this gets merged. 
#361

AdnanHodzic, 2020-02-09:
PR #361 has been merged, please pull the latest changes and retry.

asterlab, 2020-02-10:
Now it's ok thanks 

Renato Gallo

nebulastico, 2020-02-16:
Hi, can anyone tell me please how to apply the patch. Thanks

Barabazs, 2020-07-19:
Issue was resolved with PR #361

Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs