AdnanHodzic/displaylink-debian #348: HDMI Not working via usb-c hub

garrlker, 2020-01-13:
Hi, I've been trying to get 2 dongles to work for a couple days now, but not amount of trouble shooting from the ReadMe's has worked.

```
--------------- Linux system info ----------------

Distro: elementary
Release: hera
Kernel: 5.0.0-37-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: Graphics
Point-LP
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.4-1ubuntu3~18.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

The two dongles I'm trying to use are an [Anker](https://www.amazon.com/gp/product/B071G83L1J/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) and a [Novoo](https://www.amazon.com/NOVOO-Adapter-Compatible-MateBook-Chromebook/dp/B075FQY5BN)

garrlker, 2020-02-17:
It's looking like I may have been unlucky with the USB C lottery.

Those 2 dongles may be expecting there to be a displayport implementation on the port even if it doesn't have TB3, so they can claim they work with non TB3 type C ports.

My USB C port has neither of those standards and AFAIK is a plain regular USB 3.1 under the hood.

As of right now I don't think they're actually displaylink adapters.

Before this issue is closed, will someone comment a cheap USB C -> HDMI that is known to work with regular USB C ports.

Thanks in advance!

Barabazs, 2020-07-18:
It doesn't look like the installer could help you resolve your issue.

closed 2020-07-18 by Barabazs