AdnanHodzic/displaylink-debian #351: Can't detect monitors

kurojishi, 2020-01-20:
Hi,
i've been going through the throubleshooting guide

I'm on a HP 840 G6

Right now there's no, xorg.conf.d as i was iterating through possibilities, but following the post install guide I've gone through.

https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348

https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411

and still get the same, no providers. Any help will be great

```
    -------------------------------------------------------------------
     
    Starting Debug ...
     
    Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
     
    Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
     
    --------------- Linux system info ----------------
     
    Distro: Ubuntu
    Release: bionic
    Kernel: 5.3.0-26-generic
     
    ---------------- DisplayLink info ----------------
     
    Driver version: 5.1.26
    5.2.14
    DisplayLink service status: up and running
    EVDI service version: 1.6.1
     
    ------------------ Graphics card -----------------
     
    Vendor: i915
    Subsystem: [103c:8549]
    VGA: Intel Corporation Device 3ea0 (rev 02)
    VGA (3D): 
    X11 version: 1.20.5+git20191008-0ubuntu1~18.04.1
    X11 configs:
     
    -------------- DisplayLink xorg.conf -------------
     
    File: /etc/X11/xorg.conf.d/20-displaylink.conf
    cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
    Contents:
     
     
    -------------------- Monitors --------------------
     
    Providers: number : 1
    Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
     
    -------------------------------------------------------------------
     
    Use this information when submitting an issue (http://bit.ly/2GLDlpY)
```

BPagoaga, 2020-01-21:
Are you sure you are using X11 and not wayland ?
What is the output of `echo $XDG_SESSION_TYPE ` ?

piotrsikorski6, 2020-01-22:
Have that same problem, with that same symptoms. And using X11.

kurojishi, 2020-01-22:
Yep, I'm on X11. Just to be sure I'm forcing gdm too in x11

piotrsikorski6, 2020-01-22:
We should change kernel to 4.15.X?

kurojishi, 2020-01-27:
I've iterated between various tests, using the default plugin, and i'm starting to guess that's some issues with the thunderbolt port, as it don't seem to be able to get it with ```boltctl list``` even as security is disable

But I'm a bit at a loss here

kurojishi, 2020-02-05:
@piotrsikorski6 any luck with the 4.15 kernel? i didn't get a chance to try

piotrsikorski6, 2020-02-05:
@kurojishi Unfortunately - not, I came back to Windows. The crucial point of my daily work is multi screen.

kurojishi, 2020-02-05:
thanks, i'll see if i can try, right now i'm still going multiscreen through the other ports of the laptop

closed 2020-07-18 by Barabazs