AdnanHodzic/displaylink-debian #353: Flickering mouse coursor (PopOS (Ubuntu 19.10))

b3nis, 2020-01-23:
Hello,
I have two issues with displaylink-debian that is kind of weird:

1. Flickering mouse coursor sometimes
2. Computer wont startx when booting with the USB (to displaylink) connected, just end up a black screen

What can I give you to make more sense?

NGenetzky, 2020-01-30:
@b3nis , please skim through the README before posting an issue.

It states:

> Before submitting a bug report in the issue tracker, please make sure to:
>
>    read Troubleshooting most common issues.
>    when submitting a new issue, include debug information by running: `sudo ./displaylink-debian.sh --debug`

If you could post the output of that, I'm sure the developer would appreciate it.

NGenetzky, 2020-01-30:
As indicated by :+1:  above, I also have this issue.

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: eoan
Kernel: 5.3.0-26-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
NVIDIA Corporation TU106GLM [Quadro RTX 3000 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.5+git20191008-0ubuntu1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x297 cap: 0x1, Source Output crtcs: 4 outputs: 3 associated providers: 2 name:NVIDIA-0
Provider 1: id: 0x345 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x43 cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

NGenetzky, 2020-02-02:
The flickering does not appear to be present if the monitors and the USB hub are plugged in when I boot up. Figured this is good debug information or info for other people experiencing the issues.

Woifgang, 2020-04-05:
I had the same problems  with the same "20-displaylink.conf" content but another distribution -> linux mint
For me helped this one

`Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
    Option      "PageFlip" "false"
EndSection`

Barabazs, 2020-07-19:
Have you tried this [suggestion](https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only)?

Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi).

closed 2020-07-19 by Barabazs