AdnanHodzic/displaylink-debian #369: deiban 10 Lenovo L580 postinstallation does not work

smilesun, 2020-02-20:
After doing the post installation I could not get the display work 
> --------------- Linux system info ----------------
> 
> Distro: Debian
> Release: buster
> Kernel: 4.19.0-8-amd64
> 
> ---------------- DisplayLink info ----------------
> 
> Driver version: 5.2.14
> DisplayLink service status: up and running
> EVDI service version: 1.6.1
> 
> ------------------ Graphics card -----------------
> 
> Vendor: i915
> Subsystem: 620
> VGA: Intel Corporation HD Graphics 620 (rev 02)
> VGA (3D): 
> X11 version: 1.20.4-1
> X11 configs: /etc/X11/xorg.conf.d/40-libinput.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/20-intel.conf
> 
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> Contents:
>  Section "Device"
>     Identifier  "DisplayLink"
>     Driver      "modesetting"
>     Option      "PageFlip" "false"
> EndSection
> 
> -------------------- Monitors --------------------
> 
> 
> 
> -------------------------------------------------------------------
> 
> Use this information when submitting an issue (http://bit.ly/2GLDlpY)
> 
> -------------------------------------------------------------------
>

ZMcCrocklin, 2020-03-15:
I'm having the same issue after following all steps & troubleshooting in previous issue threads.  My debug output is almost the same as the OP.  My chipset is Intel 630 instead of 620, and I have 2 providers, but I disconnected all monitors except the displaylink (possibly due to 2 different config files?)

last part of my output:
```
-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x77 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 7 associated providers: 0 name:Intel
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

Yet, here is xrandr:
```
root@Revan:~# xrandr --listmonitors
Monitors: 1
 0: +*eDP1 1920/340x1080/190+0+0  eDP1
```

After a restart, here's my dmesg log:
```
root@Revan:~# dmesg | grep evdi
[  386.874208] evdi: loading out-of-tree module taints kernel.
[  386.874268] evdi: module verification failed: signature and/or required key missing - tainting kernel
[  386.876416] evdi: [I] Initialising logging on level 5
[  386.876416] evdi: [I] Atomic driver:yes
[  387.260983] evdi: [D] evdi_add_devices:174 Increasing device count to 1
[  387.261133] evdi: [D] evdi_crtc_init:416 drm_crtc_init: 0 p00000000aada7837
[  387.261140] evdi: [D] evdi_detect:95 (dev=0) poll connector state: disconnected
[  387.261151] evdi: [W] evdi_painter_send_update_ready_if_needed:550 Painter does not exist!
[  387.261270] [drm] Initialized evdi 1.6.1 20180913 for evdi.0 on minor 1
[  387.834076] evdi: [D] evdi_driver_postclose:151 (dev=0) Process tries to close us, postclose
[  387.834077] evdi: [I] Task 1025 (Xorg.wrap) of process 1025 (Xorg.wrap)
[  387.944704] evdi: [D] evdi_painter_connect:645 (dev=0) Process is trying to connect
[  387.944709] evdi: [I] Task 929 (DesktopManagerE) of process 908 (DisplayLinkMana)
[  387.944715] evdi: [D] evdi_painter_connect:695 (dev=1) Connected with 0000000034014243
[  387.944720] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  388.164548] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  388.164553] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  388.165024] evdi: [D] evdi_painter_framebuffer_size:486 Scanout buffer not set.
[  388.165026] evdi: [D] evdi_painter_mark_dirty:514 (dev=1) Skip clip rect. Scanout buffer not set.
[  388.165044] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  388.165046] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  388.220802] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
[  388.418365] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
[  388.418480] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  388.418481] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  388.418636] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  388.418637] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  391.070193] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
[  494.711533] evdi: [D] evdi_driver_postclose:151 (dev=1) Process tries to close us, postclose
[  494.711534] evdi: [I] Task 1340 (Xorg.wrap) of process 1340 (Xorg.wrap)
[  494.804015] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  494.804017] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  494.804310] evdi: [D] evdi_painter_framebuffer_size:486 Scanout buffer not set.
[  494.804311] evdi: [D] evdi_painter_mark_dirty:514 (dev=1) Skip clip rect. Scanout buffer not set.
[  494.804320] evdi: [D] evdi_detect:91 (dev=1) poll connector state: connected
[  494.804321] evdi: [D] evdi_painter_get_edid_copy:240 (dev=1) EDID valid
[  494.816872] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
[  494.952194] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
[  497.411416] evdi: [D] evdi_painter_dpms_notify:560 (dev=1) Notifying dpms mode: 3
```

moosbaue, 2020-07-09:
Hi

1st of all I want to thank Adnan for this
I had all what you describe but now it works for me (Debian 10.4)
in /etc/gdm3/daemon.conf:
\# GDM configuration storage
\#
\# See /usr/share/gdm/gdm.schemas for a list of available options.

[daemon]
\# Uncomment the line below to force the login screen to use Xorg
WaylandEnable=false

\# Enabling automatic login
\#  AutomaticLoginEnable = true
\#  AutomaticLogin = user1
.....

My /etc/X11/xorg.conf:
`Section "Device"
	Identifier "DisplayLink"
	Driver "modesetting"
\#	Option "PageFlip" "false"
EndSection
`

/etc/X11/xorg.conf.d/20-displaylink.conf:
`Section "Device"
    Identifier  "DisplayLink"
    Driver	"modesetting"
    Option	"PageFlip" "false
EndSection
`
xrandr --listproviders
`Providers: number : 5
Provider 0: id: 0xf9 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xb2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x90 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
`

I assume DisplaLink is  id: 0x41 
and so: xrandr --setprovideroutputsource 4 0

After that I have to restart dlm.service `sudo systemctl restart displaylink-driver.service` (weird but works) and voila

Barabazs, 2020-07-19:
Closing this issue due to inactivity of OP.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs