AdnanHodzic/displaylink-debian #372: Debian Testing: Some errors when execute the displaylink-debian.sh installer.

alu18999959g, 2020-03-09:
Hello ... when i running the installer. 

```
git clone https://github.com/AdnanHodzic/displaylink-debian.git
cd displaylink-debian/ && sudo ./displaylink-debian.sh
```
Some errors occur:

> patching file .travis.yml
> Hunk #1 FAILED at 26.
> 1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
> patching file Makefile
> Reversed (or previously applied) patch detected!  Skipping patch.
> 1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
> patching file library/Makefile
> Hunk #1 FAILED at 2.
> 1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
> patching file module/Makefile
> Hunk #1 FAILED at 43.
> 1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
> patching file module/evdi_drv.c
> Hunk #1 FAILED at 81.
> 1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
> patching file module/evdi_drv.h
> Hunk #1 FAILED at 20.
> Hunk #2 FAILED at 51.
> Hunk #3 FAILED at 112.
> 3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
> patching file module/evdi_gem.c
> Hunk #1 FAILED at 61.
> Hunk #2 FAILED at 254.
> Hunk #3 FAILED at 569.
> Hunk #4 FAILED at 581.
> 4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej

When restarting I cannot login..the only way is to delete

> /etc/X11/xorg.conf.d /20-displaylink.conf

After login:

```
sudo ./displaylink-debian.sh --debug
[sudo] password for hector: 
```

> 
> -------------------------------------------------------------------
> 
> Starting Debug ...
> 
> Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
> 
> Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
> 
> --------------- Linux system info ----------------
> 
> Distro: Debian
> Release: bullseye
> Kernel: 5.4.0-4-amd64
> 
> ---------------- DisplayLink info ----------------
> 
> Driver version: 5.2.14
> DisplayLink service status: up and running
> EVDI service version: 1.6.1
> 
> ------------------ Graphics card -----------------
> 
> Vendor: i915
> Subsystem: 4th
> VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
> VGA (3D): NVIDIA Corporation GM108M [GeForce 930M] (rev a2)
> X11 version: 1.20.7-4
> X11 configs: /etc/X11/xorg.conf.d/20-nvidia.conf /etc/X11/xorg.conf
> 
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No existe el fichero o el directorio
> Contents:
>  
> 
> -------------------- Monitors --------------------
> 
> Providers: number : 2
> Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
> Provider 1: id: 0x1de cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
> 
> -------------------------------------------------------------------
> 

I have a laptop with nvidia optimus(Use NVIDIA graphics only ). My setup is:

`/etc/X11/xorg.conf`

> Section "Module"
>     Load "modesetting"
> EndSection
> 
> Section "Device"
>     Identifier "nvidia"
>     Driver "nvidia"
>     BusID "PCI:1:0:0"
>     Option "AllowEmptyInitialConfiguration"
> EndSection
> 

https://download.nvidia.com/XFree86/Linux-x86_64/440.59/README/randr14.html

`/usr/share/sddm/scripts/Xsetup`

> #!/bin/sh
> # Xsetup - run as root before the login dialog appears
> xrandr --setprovideroutputsource modesetting NVIDIA-0
> xrandr --auto
> xrandr --dpi 96

Any suggestions ?

Thanks in advance.

Erwin-D, 2020-04-29:
Found a workaround and described the steps here: #398

Barabazs, 2020-07-19:
Issue has been resolved since then.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs