AdnanHodzic/displaylink-debian #373: Looking for maintainers (developers)

AdnanHodzic, 2020-03-11:
As stated in README, I'm looking for maintainers (developers) to continue work on this project.

Your duties would be:

* displaylink-debian tool maintenance and further development
* MR (merge requests) code review and approval
* Bug/issue resolution and triage

Please inform me if you have any additional questions and/or comments.

Adnan

Ramesh534, 2020-03-11:
interested

AdnanHodzic, 2020-03-12:
HI @Ramesh534, I think ideal start for you would be [to look at bug reports](https://github.com/AdnanHodzic/displaylink-debian/issues), as this is something that's been neglected the most.

A lot of them are low hanging fruit and could easily be closed. This will allow you to learn about the project and its code. After that we can discuss rest of the points/duties I mentioned here.

In meantime, please inform me if you have any questions and/or comments.

tedce, 2020-05-21:
I would be interested in helping with this project! This has been so useful for me and coworkers' workstations and I'd love to be a part of it.

AdnanHodzic, 2020-05-23:
> HI @Ramesh534, I think ideal start for you would be [to look at bug reports](https://github.com/AdnanHodzic/displaylink-debian/issues), as this is something that's been neglected the most.
>
> A lot of [...]

Hi @tedce that's great to hear. I think same thing which I said here applies for you as well. After you go bit acquainted best start would be to start by submitting PR for either reported bug issues/fixed or improvements.

RameshPishini, 2020-05-23:
Hi, @AdnanHodzic last week I was quite busy with other works. Let me have a look into a few bugs and also please share your availability next week so that we can schedule a call to discuss the project.

AdnanHodzic, 2020-06-05:
Hi @RameshPishini I hope the amount of time it took me to reply back illustrates how busy I am on my side as well :)

If you've resolved any bugs with code changes, please feel free to create a PR and I'll do my best to further assist you during code review process.

Regarding a call you suggested, at this point I would prefer emails. Hence please feel free to ask me about anything via email.

Ramesh534, 2020-06-14:
@AdnanHodzic unable to push the modification in github. Getting permission denied.

AdnanHodzic, 2020-06-14:
@Ramesh534 you need to [create a merge request](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request).

Barabas-dev, 2020-07-17:
Hi AdnanHodzic. I can help with cleaning up old issues and merging or referencing duplicates.
Let me know if you're interested.

AdnanHodzic, 2020-07-18:
@Barabas-dev yes, I would definitely be interested. Looking forward to your MR!

Barabas-dev, 2020-07-18:
@AdnanHodzic how can we proceed for issue/bug triage. If you want, we can mail, chat or call about this.

PS we talked on LinkedIn a while ago. (Erwin)

AdnanHodzic, 2020-07-18:
@Barabazs I remember, glad I hear from you again :)

Basically same thing I said to @RameshPishini above still stands.

> HI @Ramesh534, I think ideal start for you would be [to look at bug reports](https://github.com/AdnanHodzic/displaylink-debian/issues), as this is something that's been neglected the most.
>
> A lot of [...]

And I suggested to him, at this point I would prefer if we communicated via email/chat regarding this topic.

Barabazs, 2020-07-18:
@AdnanHodzic Unfortunately there are no options to allow someone to close/label issues without giving them [collaborator privileges.](https://docs.github.com/en/github/setting-up-and-managing-your-github-user-account/permission-levels-for-a-user-account-repository)

AdnanHodzic, 2020-07-18:
@Barabazs sent collaborator invite.

Ramesh534, 2020-07-18:
@AdnanHodzic sincere apologies at this moment I am overloaded with my personal and professional work. That's why I am not able to work on it.

AdnanHodzic, 2020-07-18:
@Ramesh534 hey, no worries! You're always welcome to join if this situation changes for you.

Barabazs, 2020-07-18:
> @Barabazs sent collaborator invite.

Thank you. I hope you don't mind it if I close old/inactive issues in a "firm" manner.

AdnanHodzic, 2020-07-18:
@Barabazs yes, I've see what you're doing and you have my full support. Please proceed :)

xAlimorAx, 2021-03-02:
Hey Guys,

   First off, Great work and I don't know if anyone has told you guys yet, you are all amazing and we appreciate what your doing!
So now to the point, I am not much of a dev... But if you need any help with public relations or beta testing quickly, & efficiently... Anything to help! I would love to get involved somehow. I handle myself in Kali Linux, I have a tri-boot setup using Grub as my BL. I run Win10/Kali/Hackintosh. I can do some graphic design, I'm good with word, excel, & a few others. I've done some web-design and database work. Other than that... I am a pretty quick learner, I'm starting to dabble with Python/Raptor to get my toes wet.

  Anyway if you can use me in any way, please let me know! If not, i will just try to keep supporting and submitting useful debugs or whatever else I can.

Thanks again,
AlimorA

AdnanHodzic, 2021-03-11:
Hi @xAlimorAx, first of all I apologize for such delayed reply. Really glad to hear you're willing help, much appreciated!

@Barabazs if you need help, please consider involving @xAlimorAx with some of the tasks.

xAlimorAx, 2021-03-11:
It’s ok, thanks for responding and yes I’m ready and willing to do whatever I can to help.

Barabazs, 2021-03-13:
Thank you for your support! I'll be sure to ping you when necessary.

m0lese, 2021-08-03:
I'm interested since the project is mainly on shell script.

I've looked at [displaylink-debian.sh](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh) & [displaylink.sh](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink.sh) and invoked [shellcheck](https://www.shellcheck.net/) on them. A lot of bugs were detected by [shellcheck](https://www.shellcheck.net/) which should be fixed and pretty easy to do. The code is also very messy, there's a tool called `shfmt` in [mvdan/sh](https://github.com/mvdan/sh) which can be used to format the shell scripts.

artukryp, 2022-02-05:
Hi, I'm interested

too-difficult-to-name, 2022-03-22:
> the

wow, nice!

scgoeswild, 2022-11-21:
Hi @AdnanHodzic 
I am sw eng currently managing some kali linuxes with display port. 
I would like to contribute on your repo. 

Simone

AdnanHodzic, 2022-11-24:
@scgoeswild tagging @Barabazs as he's now the maintainer of this project.