AdnanHodzic/displaylink-debian #376: Re/Uninstall looks for displaylink-installer.sh

robotard, 2020-03-19:
I've recently seen some issue on reinstalling ( I think uninstall too ) whereby it's looking for the following path:

/opt/displaylink/displaylink-installer.sh

Strangely if you copy the contents from the 5.2 folder there, and uninstall, it clears up this folder, but am guessing somewhere in the process this is not getting copied across... Not sure under which circumstances it's not copied as obvs am running as sudo, but have seen this occuring,,,

robotard, 2020-03-19:
```
Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: No such file or directory

-------------------------------------------------------------------

Performing clean-up
```

AdnanHodzic, 2020-03-20:
I believe this was fixed as part of your #377 PR

closed 2020-03-20 by AdnanHodzic

robotard, 2020-04-01:
It does appear so. :) 

Not entirely sure as to why that my little fix cascaded down to that other one being fixed...