AdnanHodzic/displaylink-debian #378: Failed patching with latest evdi

kalon33, 2020-03-20:
Using latest evdi, the patch doesn't apply properly anymore:

```
Clonage dans 'evdi'...
remote: Enumerating objects: 187, done.
remote: Counting objects: 100% (187/187), done.
remote: Compressing objects: 100% (107/107), done.
remote: Total 1338 (delta 118), reused 141 (delta 80), pack-reused 1151
Réception d'objets: 100% (1338/1338), 804.40 KiB | 1.40 MiB/s, fait.
Résolution des deltas: 100% (887/887), fait.
--2020-03-20 12:24:08--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Résolution de crazy.dev.frugalware.org (crazy.dev.frugalware.org)… 85.10.194.195
Connexion à crazy.dev.frugalware.org (crazy.dev.frugalware.org)|85.10.194.195|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 6947 (6,8K) [text/plain]
Enregistre : «evdi-all-in-one-fixes.patch»

evdi-all-in-one-fixes.patch                        100%[===============================================================================================================>]   6,78K  --.-KB/s    ds 0,003s  

2020-03-20 12:24:09 (2,58 MB/s) - «evdi-all-in-one-fixes.patch» enregistré [6947/6947]

patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej

```

persistentpenguin, 2020-04-17:
Same here. The whole setup fails now and cannot proceed. Please fix.

AdnanHodzic, 2020-04-21:
Change needs to be [done on this line](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L21). 

Considering what I [stated here](https://github.com/AdnanHodzic/displaylink-debian#looking-for-maintainers-developers). I would prefer if someone addressed this issue and created a PR which I would be glad to review and merge.

Erwin-D, 2020-04-29:
Found a workaround and described the steps here: #398

Barabazs, 2020-07-19:
Issue has been resolved.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs