AdnanHodzic/displaylink-debian #380: Uninstall failure -  `bash: /opt/displaylink/displaylink-installer.sh: No such file or directory`

robotard, 2020-03-22:
Nice and easy one in such that I've pretty much isolated the problem and coded a fix (read below though pls before I push)

There's a bug preventing uninstallation (and I think reinstallation too), not finding "displaylink-installer.sh", which appears to not actually be copied to "/opt/diplaylink" during installation... You see the error below:

`bash: /opt/displaylink/displaylink-installer.sh: No such file or directory`

Interestingly, if you manually copy it there (or use the script below that I knocked together), then when uninstallation from our script has completed, it has removed the "/opt/displaylink" folder as part of cleanup...

Ok, so I've put together this little script (makeopt.sh) to get around the issue to allow me to fully test instal/uninstall in general... 
```
#!/bin/bash

#./5.2/displaylink-driver-5.2

# Quick hardcoded fix for displaylink-debian uninstallation failure
# Stick this in the displaylink-debian folder
# It assumes that we have a 5.2 folder in there (can get from installation)

driverfolder="./5.2/displaylink-driver-5.2/"
#I've copied the current version folder to the below for my testing
#driverfolder="./5.2bak/displaylink-driver-5.2/"
echo "$driverfolder"
optfolder="/opt/displaylink"

if [ -d  $driverfolder ]
then
	echo "Exists, lets copy to $optfolder"
	#check if folder exists
	if [ ! -d $optfolder ]
	then
		mkdir $optfolder
		echo "created $optfolder"
	fi
	cp -r $driverfolder* $optfolder
	echo "copied files"
	ls $optfolder
else
	echo "Oh really, no driver folder?"
fi
```
Now this works just fine, but I've tried implementing with our current $driver_dir into the post_install() method and grabbing the correct folder for the current version that we have downloaded (see path below), but for some strange reason my if [ -d ] check on this folder seems to not find the folder. I've even tried moving it to the top of post_install()

"$driver_dir/displaylink-driver-${version}" 

Am sure there is something very tiny that I'm missing, and as I said, it addresses the above bug. Am more than happy to push the version of displaylink-debian.sh with the "fix" commented out, and then you'll only have to mend the folder path check?

It's an odd one, but don't want to open a new bug when the premise fix in my code just needs a little nudge in the right direction ;)

Let me know what your thoughts are...

MaiKuraki, 2020-05-12:
thanks! I got this problem, and your solution works well!

robotard, 2020-05-12:
You're very welcome... Adnan et al definitely did the main brunt of the work, so massive kudos to them too :)

closed 2020-10-16 by Barabazs