AdnanHodzic/displaylink-debian #382: Just cant get it to work.

junkergastabud, 2020-04-02:
Been trying for weeks to get my laptop, a Lenovo 720s to work with thunderbolt docking and using an external hdmi screen. 

it just doesn't detect any external screens

Plz help. Would be nice to work from home on a screen bigger than 13" ;)
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-18-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: 
Subsystem: 620
[1180]:
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.7-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Erwin-D, 2020-04-30:
Did you run into errors while installing?

Barabazs, 2020-07-19:
You could try using this configuration:

```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs