AdnanHodzic/displaylink-debian #383: My vga to usb adapter not working

waltim, 2020-04-08:
My vga to usb adapter is not recognized to displaylink i think.

The output of command "sudo ./displaylink-debian.sh --debug" is:

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.0.0-23-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 3rd
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): NVIDIA Corporation GK107M [GeForce GT 740M] (rev a1)
X11 version: 1.19.6-1ubuntu4.4

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Arquivo ou diretório inexistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x66 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0x3f cap: 0x4, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:nouveau

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Someone would like to help me?

waltim, 2020-04-13:
I managed to make it work on my machine, but using windows 10. I had to install a drive called FL2000 by Fresco Logic.

The device I'm using is:

Lotus 3.0 to Vga Converter Cable - Full Hd from Lotus

Would someone like to help me?

Barabazs, 2020-07-19:
This isn't a DisplayLink device right?

closed 2020-07-29 by Barabazs