AdnanHodzic/displaylink-debian #384: Troubles with Debian 10

renierts, 2020-04-09:
Hi,

I have the problem that no monitors are recognized. 

The installation was finished. However, while rebooting, the system is waiting for a long time, and the following output is printed until approx. 3min:

"A start job is running for Helper to synchronize boot up for ifupdown (44s / 3min)"

Here is the output of "sudo ./displaylink-debian.sh --debug"

<pre>-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): NVIDIA Corporation GM107M [GeForce GTX 850M] (rev a2)
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section &quot;ServerLayout&quot;
    Identifier &quot;layout&quot;
    Screen 0 &quot;nvidia&quot;
    Inactive &quot;intel&quot;
EndSection

Section &quot;Device&quot;
    Identifier &quot;intel&quot;
    Driver &quot;modesetting&quot;
    Option &quot;AccelMethod&quot; &quot;None&quot;
EndSection

Section &quot;Screen&quot;
    Identifier &quot;intel&quot;
    Device &quot;intel&quot;
EndSection

Section &quot;Device&quot;
    Identifier &quot;nvidia&quot;
    Driver &quot;nvidia&quot;
    Option &quot;ConstrainCursor&quot; &quot;off&quot;
EndSection

Section &quot;Screen&quot;
    Identifier &quot;nvidia&quot;
    Device &quot;nvidia&quot;
    Option &quot;AllowEmptyInitialConfiguration&quot; &quot;on&quot;
    Option &quot;IgnoreDisplayDevices&quot; &quot;CRT&quot;
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
</pre>

asdfasdf

mikethebos, 2020-06-05:
Run `apt remove ifupdown`. It's safe.
For HD4000 iGPU, I had to put this at /etc/X11/xorg.conf.d/20-displaylink.conf:
````
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection
````
Then reboot and run `xrandr --setprovideroutputsource 4 0` until your DE's display settings show a new screen. '4' may not be your provider. Try numbers 1 through 9.

Barabazs, 2020-07-19:
@renierts Did the suggestion by @mikethebos resolve your issue?

renierts, 2020-07-19:
The suggestion did not resolve my issue. Sorry for not replying. Anyway, as I don't have the product anymore, I would close this issue.

closed 2020-07-19 by renierts