AdnanHodzic/displaylink-debian #388: Poor performance when using DL-5500-based dongle + Intel integrated graphics

aph3rson, 2020-04-16:
I'm using a Thinkpad T450 with Kali Rolling. I have three Dell U2515H-series monitors - two are connected to the Ultra dock for the laptop, and a third through a Startech USB32DP4K dongle.

I'm able to install the drivers just fine, but I'm running in to some major performance issues. It appears that the changes to xorg.conf make my non-dongle'd displays use a modesetting driver, rather than the Intel driver. I've tried a few different combinations, but I'm thinking my grasp of X.Org's configuration just isn't there.

Here's what I currently have in my configuration file:
```text
root@laptop:~# cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

This gives my two non-dongled monitors full performance, but X11 doesn't pick up the dongle'd GPU:
```text
root@laptop:~# xrandr --listproviders 
Providers: number : 1
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel
```

If I remove the `Device` section, I'm able to use all three monitors, but only on the modesetting driver (as confirmed by `xrandr` again), and performance is severely impacted. These are three 1440p displays, so it's already pushing the limits of the available GPU power on this device...

Any ideas?

aph3rson, 2020-04-16:
A lot of this is duplicative, but here's debug output anyways:

```text
root@laptop:/opt/displaylink-debian# ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.4.0-kali4-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.20.8-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

Barabazs, 2020-07-19:
I'm not sure if this is related to the use of the installer.
Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi/).

closed 2020-07-29 by Barabazs