AdnanHodzic/displaylink-debian #389: Request for support, Pop_OS please...

fordjonathan, 2020-04-23:
Good Evening, I'm requesting support be rolled in for Pop_OS 20.04 please, I'm trying to modify the script myself at the moment though, but not with much success...

Many thanks in advance
Jonathan

fordjonathan, 2020-04-23:
Not to worry, I modified the script successfully, I just had to fiddle the formatting, now how do I upload the changes...?

AdnanHodzic, 2020-04-23:
Great, looking forward to your MR then, you can follow this [link how to create one](https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request).

JulienOpto, 2020-04-26:
Hi ! I also need Pop_OS 20.04 support. I manage to modify the script in order to get pop focal repos but that still doesn't work for me. Here is the debug :

`sudo ./displaylink-debian.sh --debug             ─╯
[sudo] password for julienmartin: 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: focal
Kernel: 5.4.0-7625-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

`
and now xrandr: 

`xrandr --listproviders                                                                  ─╯
Providers: number : 1
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:modesetting
`
I tried all suggestions mentionned in the post installation guide but nothing worked so far.
Can someone please help me ?

closed 2020-05-07 by fordjonathan