AdnanHodzic/displaylink-debian #390: paypal donation link does not work

polrus, 2020-04-24:
Hi
Wanted to donate but the link fails

AdnanHodzic, 2020-04-25:
Hi, thank you for your support, much appreciated.

I've [updated the PayPal link](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md#donate-paypal-or-bitcoin), please try again.

AdnanHodzic, 2020-05-02:
@polrus I've updated the link again, I kept forcing it to be in English locally which made it expire each time.

Could you [please try with this updated one](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=adnan%40hodzic.org&item_name=Contribution+for+work+on+debian-displaylink&currency_code=EUR&source=url)?

I've also updated the [link in README](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md#donate-paypal-or-bitcoin).

Thanks

Barabazs, 2020-07-18:
I can confirm that the link works.

PS: please don't be offended by the amount.

closed 2020-07-18 by Barabazs