AdnanHodzic/displaylink-debian #391: I can't install it on Ubuntu 20.04

chilcano, 2020-04-24:
I tried the DisplayLink 5.2 drivers in Ubuntu 18.04 and worked, after I tried same drivers in Ubuntu 20.04 and I got this error:
```
...
Distribution discovered: Ubuntu 20.04 LTS
Installing
Configuring EVDI DKMS module
Registering EVDI kernel module with DKMS
Building EVDI kernel module with DKMS
ERROR (code 3): Failed to build evdi/5.2.14. Consult /var/lib/dkms/evdi/5.2.14/build/make.log for details..
``` 

After that I tried DisplayLink-Debian without success and getting similar error:
```
...
Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------
./displaylink-debian.sh: line 405: ./evdi.sh: No such file or directory
``` 

Kernels versions:
``` 
// Ubuntu 20.04 Focal Fossa
$ uname -r
5.4.0-26-generic

// Ubuntu 18.04
$ uname -r
5.3.0-46-generic
```

Any idea is welcome.
Regards.

xdontkillmeisuck, 2020-04-25:
Same!

Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------
./displaylink-debian.sh: line 405: ./evdi.sh: No such file or directory

5.4.0-7625-generic #29~1587437458~20.04~2960161~dev-Ubuntu SMP Tue Apr 21 05:18

BellaMay95, 2020-04-25:
I also saw the same issue.
Problem was that I used method #2 and just did a "wget" for the root script file...the repo also has the ./evdi.sh that it thinks is missing.
I just used git to clone the repo and run the script without issues.

README should be updated for method #2 to also download the other two .sh files in the repo.

chilcano, 2020-04-27:
It worked using 2nd method.
Important: connect the external screen after installation and after rebooting.

xdontkillmeisuck, 2020-04-27:
I was able to compile install installed correctly, but it caused tearing and weird artifact in electron apps, but since I have a system76 system they use a modified kernel so not your problem.

Cheers,
Jeff

RidgeLock, 2020-04-28:
Now i got another error: 

can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/.travis.yml b/.travis.yml
|index 0a9c52b..932d18f 100644
|--- a/.travis.yml
|+++ b/.travis.yml
--------------------------
File to patch:

chilcano, 2020-04-28:
Hi @RidgeLock 
just follow the 2nd method explained here: https://github.com/AdnanHodzic/displaylink-debian#download-and-run-script-without-git-method-2

If so, you have wget 3 bash scripts and run the installer:
```
$ wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh 
$ wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh 
$ wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/evdi.sh
$ chmod +x displaylink-debian.sh evdi.sh && sudo ./displaylink-debian.sh
```
Hope it helps you.

MelvinLoos, 2020-04-28:
> Hi @RidgeLock
> just follow the 2nd method explained here: https://github.com/AdnanHodzic/displaylink-debian#download-and-run-script-without-git-method-2
>
> If so, you have wget 3 bash scripts and run [...]

Why specifically method 2? The only difference I see is using git right?

jpmti2016, 2020-04-29:
Fail got:
...
Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  

-------------------------------------------------------------------
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libdrm-dev is already the newest version (2.4.101-2).
libelf-dev is already the newest version (0.176-1.1build1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'evdi'...
remote: Enumerating objects: 424, done.
remote: Counting objects: 100% (424/424), done.
remote: Compressing objects: 100% (196/196), done.
remote: Total 1575 (delta 280), reused 350 (delta 225), pack-reused 1151
Receiving objects: 100% (1575/1575), 888.51 KiB | 3.24 MiB/s, done.
Resolving deltas: 100% (1049/1049), done.
--2020-04-29 09:53:52--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Resolving crazy.dev.frugalware.org (crazy.dev.frugalware.org)... 85.10.194.195
Connecting to crazy.dev.frugalware.org (crazy.dev.frugalware.org)|85.10.194.195|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6947 (6.8K) [text/plain]
Saving to: ‘evdi-all-in-one-fixes.patch’

evdi-all-in-one-fixes.patch                        100%[===============================================================================================================>]   6.78K  --.-KB/s    in 0s      

2020-04-29 09:53:52 (65.3 MB/s) - ‘evdi-all-in-one-fixes.patch’ saved [6947/6947]

./evdi.sh: line 7: $patchPath: ambiguous redirect
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Entering directory '/home/front-end/Desktop/Manual Display Link/displaylink-debian-master/evdi/module'
make -C /lib/modules/5.4.0-28-generic/build M=$PWD
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
arch/x86/Makefile:147: CONFIG_X86_X32 enabled but no binutils support
make[2]: *** No rule to make target 'Display'.  Stop.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make[1]: *** [Makefile:71: module] Error 2
make[1]: Leaving directory '/home/front-end/Desktop/Manual Display Link/displaylink-debian-master/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat '../library/libevdi.so.1.*': No such file or directory

errst, 2020-04-29:
@chilcano Can you post your **/etc/X11/xorg.conf.d/20-displaylink.conf** file content and also -if possible- debug output? I've Ubuntu 20.04 LTS and installed driver as I explained in [here](https://github.com/AdnanHodzic/displaylink-debian/issues/396#issuecomment-621487771) but no lock. Thanks in advance.

shawnGao87, 2020-04-30:
I had the same issues. Just got some system updates in 10 minutes ago. And then I was able to install the driver successfully. Although, it doesn't seem to work - the monitors are still not being recognized. I have Dell D3100. Debug info below:

`Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
VGA (3D): NVIDIA Corporation TU117GLM [Quadro T2000 Mobile / Max-Q] (rev a1)
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x218 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x23f cap: 0x2, Sink Output crtcs: 3 outputs: 4 associated providers: 1 name:modesetting

`

chilcano, 2020-04-30:
Hi @errst 
Below my debug info.
As I stated before, I've noticed an [annoying behavior when rebooting while plugged DisplayLink screen](https://github.com/AdnanHodzic/displaylink-debian/issues/395), other thing I've noticed is the fan is running more time than when I had Ubuntu 18.04.
I hope it helps.
Regards.
```
--------------- Linux system info ----------------
Distro: Ubuntu
Release: focal
Kernel: 5.4.0-26-generic
---------------- DisplayLink info ----------------
Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1
------------------ Graphics card -----------------
Vendor: i915
Subsystem: 3rd
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
-------------------- Monitors --------------------
Providers: number : 2
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 1 name:Intel
Provider 1: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
-------------------------------------------------------------------
```

cpieroni, 2020-04-30:
It doesn't work for me either on Ubuntu 20.04.
@chilcano rebooting or changing USB port didn't work for me.

Below my debug info:

`--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): NVIDIA Corporation GK208BM [GeForce 920M] (rev a1)
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x278 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x29f cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:Intel

-------------------------------------------------------------------
`

errst, 2020-04-30:
@chilcano Thanks for sharing. I've tried things you've said and same config that you have but nope, it didn't work for me. I literally hate displaylink.

errst, 2020-04-30:
Last commits solved the issue. It is working now, fyi.

chilcano, 2020-04-30:
Good to know it is working @errst !!
Ah, what do you mean with "last commits"?.
Regards

Joeb3219, 2020-05-01:
@chilcano At least on my end, commit `31fa071f270cc0bd42d0a82c66096afc69f0f194` seems to work fine, whereas two days ago this wasn't working on my end either for Ubuntu 20.04.

chilcano, 2020-05-02:
Thanks @Joeb3219 !

wsadf, 2020-05-04:
Acabei de tentar instalar no Ubuntu 20.04 e não funcionou, como vocês conseguiram? qual o link deste commit 31fa071f270cc0bd42d0a82c66096afc69f0f194?

wsadf, 2020-05-04:
> @chilcano Pelo menos do meu lado, o commit `31fa071f270cc0bd42d0a82c66096afc69f0f194`parece funcionar bem, enquanto dois dias atrás isso também não estava funcionando no Ubuntu 20.04.

Conseguiu fazer funcionar o displayLink?

chilcano, 2020-07-20:
No @Barabazs 
Thanks.

closed 2020-07-20 by chilcano

wsadf, 2020-07-20:
Depois dessa atualização, o displayLink funcionou: https://www.displaylink.com/downloads/ubuntu