AdnanHodzic/displaylink-debian #393: Performance Issue when internal Monitor is disabled

moritzknorr, 2020-04-26:
Hello, I am getting low performance on my external monitor, when my internal monitor is disabled. (Window moving, hovering effects) The mouse is as responsive as always.
UEFI/Secure Boot is disabled; No issues during the installation, except for Cursor artifacts, that were gone after another reboot.
`-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-26-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------`

romgere, 2020-04-30:
Same issue here :disappointed:

gat-hobort, 2020-05-02:
Had same issue (ubuntu 20.04)
but found a quick and dirty workaround, here it is:

1.  with no displaylink connected identify your laptop display name like this:

`xrandr --query | grep " connected"`

for me the output was:

`eDP-1 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 293mm x 165mm`

2. now connect the displaylink dock

3. open a terminal a prepare a command like this:

`xrandr --output eDP-1 --mode 1920x1080 --pos 0x0 --rotate normal`

where you use the output name and mode from the first query output
do not execute that, just have it ready for run

4. close the lid and verify that performance issues preset (try to move the terminal window)

5. in the terminal window now run the prepared command (press enter)
everything should work normally again, no performance issues with lid closed

6. if that command solved the problem prepare a script which runs automatically when lid is closed, for example: /usr/local/bin/displaylink-lid-fix.sh with this content:

```
#!/bin/bash

NAME=eDP-1
MODE=1920x1080
MYUSER=johndoe

grep -q closed /proc/acpi/button/lid/LID/state

if [ $? = 0 ]; then
# lid has been just closed

if [ $UID -eq 0 ]; then
    # run by root (triggered by lid closing)

    DISPLAY=:0.0 su $MYUSER -c "xrandr --output ${NAME} --mode ${MODE} --pos 0x0 --rotate normal"

else
    # run by user manually
    xrandr --output ${NAME} --mode ${MODE} --pos 0x0 --rotate normal
fi

fi
```

replace NAME MODE and MYUSER with your values.
Make it executable

7. create a /etc/acpi/event/laptop-lid file with this content:

```
event=button/lid.*
action=/usr/local/bin/displaylink-lid-fix.sh
```

8. reboot

asenchristov, 2020-05-02:
same issue here:

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: HD
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 2 name:modesetting
Provider 1: id: 0x10d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

```

kuateric, 2020-05-04:
> Had same issue (ubuntu 20.04)
> but found a quick and dirty workaround, here it is:
> 
> 1. with no displaylink connected identify your laptop display name like this:
> 
> `xrandr --query | grep " connected"`
> 
> for me the output was:
> 
> `eDP-1 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 293mm x 165mm`
> 
> 1. now connect the displaylink dock
> 2. open a terminal a prepare a command like this:
> 
> `xrandr --output eDP-1 --mode 1920x1080 --pos 0x0 --rotate normal`
> 
> where you use the output name and mode from the first query output
> do not execute that, just have it ready for run
> 
> 1. close the lid and verify that performance issues preset (try to move the terminal window)
> 2. in the terminal window now run the prepared command (press enter)
>    everything should work normally again, no performance issues with lid closed
> 3. if that command solved the problem prepare a script which runs automatically when lid is closed, for example: /usr/local/bin/displaylink-lid-fix.sh with this content:
> 
> ```
> #!/bin/bash
> 
> NAME=eDP-1
> MODE=1920x1080
> MYUSER=johndoe
> 
> grep -q closed /proc/acpi/button/lid/LID/state
> 
> if [ $? = 0 ]; then
> # lid has been just closed
> 
> if [ $UID -eq 0 ]; then
>     # run by root (triggered by lid closing)
> 
>     DISPLAY=:0.0 su $MYUSER -c "xrandr --output ${NAME} --mode ${MODE} --pos 0x0 --rotate normal"
> 
> else
>     # run by user manually
>     xrandr --output ${NAME} --mode ${MODE} --pos 0x0 --rotate normal
> fi
> 
> fi
> ```
> 
> replace NAME MODE and MYUSER with your values.
> Make it executable
> 
> 1. create a /etc/acpi/event/laptop-lid file with this content:
> 
> ```
> event=button/lid.*
> action=/usr/local/bin/displaylink-lid-fix.sh
> ```
> 
> 1. reboot

Thank you for your fix. it works fine. But when I reboot I get the same performance problem again. Any idea how can i fix that ?

moritzknorr, 2020-05-04:
Hey asenchristov, thanks for your reply.

This simply enables the internal output and therefore the performance issues are gone.
> `xrandr --output eDP-1 --mode 1920x1080 --pos 0x0 --rotate normal`
I have a 37" widescreen monitor and I do not want my notebook to output anything (esp if closed).

**Does anybody know why there are performance issues when the internal display is deactivated?**

Cheers.
Moritz

gat-hobort, 2020-05-04:
Hi moritzknorr

at the first sight it enables the internal output. But in reality (for me) it is not outputting anything.
Tested with me lenovo X390 - where the lid close event triggers when there is a gap about 1cm so you can see your screen switching off.

After that xrandr it does NOT switch back.

Another "proof" (again for me, please test yourself) is that the ubuntu screen display settings is not showing the internal monitor as active, and you cannot move any window outside your external monitor(s)

I am not saying this is the proper solution to the problem - but until a proper fix is available I can live with that

@kuateric: I would say after reboot open/close the lid, or start the script manually

moritzknorr, 2020-05-04:
Hey @gat-hobort,

thanks for your reply. I tested what you said. You are absolutely right!
Nevertheless this seems a bit "dirty" to me.

Cheers.
Moritz

kuateric, 2020-05-04:
Thank you @gat-hobort, I tested both solutions they work. The most easy way is to open and close the lid after the reboot.

eyalmazuz, 2020-05-20:
> Had same issue (ubuntu 20.04)
> but found a quick and dirty workaround, here it is:
>
>     1. with no displaylink connected identify your laptop display name like this:
>
>
> `xrandr --query | grep " [...]

Hi, I tried that and it works but for some reason, the displaylink-lid-fix doesn't run when I close the lid, I tried looking on the internet why, some people said the displaylink-lid-fix.sh need to be named lid.sh and place at /etc/acpi and the laptop-lid need to be named lm_lid.
but it still doesn't work, I tried running the script alone without the laptop_lid/lm_lid event and the script works fine.
Any idea how can I fix that?

matthew-davis, 2020-05-21:
I would like to add my details to the list as this is the same behaviour on my system:

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-31-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x162 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x140 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x11e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xe9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

I can also report that the workaround script works, however like eyalmazuz I cannot get the bash script to run on the lid close. I did notice that of the events listed in the /etc/acpi folder there doesn't seem to be one that interfaces with the lid and also I didn't have an "/etc/acip/event" folder only an "/etc/acpi/events" notice the "s" perhaps my computer doesn't capture the lid event through acpi? It's a Dell XPS 13. Doing research on google I found a few explanations of how to capture the lid opening and closing, but like the script above they all presuppose that acpi has a captured event for the lid. If anyone has any ideas I'd love to give it a go, until then I'm typing a few things in each time I go to external monitors.

eyalmazuz, 2020-05-23:
> I would like to add my details to the list as this is the same behaviour on my system:
>
> --------------- Linux system info ----------------
>
> Distro: Ubuntu
> Release: focal
> Kernel: 5.4.0-31-generic [...]

I did a workaround for the lid close thing, it's not optimal but it's easier.
I've binded the script in /usr/local/bin/displaylink-lid-fix.sh with a keyboard shortcut (mine is Super+F12) so I just press those every time I close the lid.
to do so open the keyboard shortcut in the settings menu, go all the way down and press the '+' symbol, on Name put whatever you want, on command put bash /usr/local/bin/displaylink-lid-fix.sh and shortcut again whatever you want.

Barabazs, 2020-09-06:
The devs at Displaylink are working on a solution with xserver.
FYI:
https://www.displaylink.org/forum/showthread.php?t=67148
https://gitlab.freedesktop.org/xorg/xserver/-/issues/1028

closed 2020-10-16 by Barabazs