AdnanHodzic/displaylink-debian #395: Black DisplayLink screen after Ubuntu 20.04 Login

chilcano, 2020-04-28:
Hi there.

I've followed [2nd method](https://github.com/AdnanHodzic/displaylink-debian/issues/391) to install DisplayLink-Debian drivers and EVDI module. That worked without problems. 
But now I've noticed that if I reboot and login while keeping plugged my DisplayLink screen I got a black DisplayLink screen. 

If I unplug the DisplayLink screen, reboot, login and plug it again, the DisplayLink screen will work but loss my screen layout configuration. But If I unplug and plug it again, the Ubuntu 20.04 will reload my previous screen layout configuration.

Any idea how to solve this annoying behavior?
Regards.

Barabazs, 2020-07-18:
Loss of layout configuration can be addressed with  xrandr (or arandr if you prefer a GUI).

What happens with your laptop monitor when you reboot while having the dock plugged in?

Barabazs, 2020-07-29:
Closing this issue since it doesn't relate to issues with the installer.

closed 2020-07-29 by Barabazs