AdnanHodzic/displaylink-debian #397: Installation succesful, external monitors blank or garbage

julianandrews, 2020-04-29:
Hi!

I was able to run the install script, and the status reported is good. The monitors weren't automatically detected, but using `xrandr --setprovideroutputsource` I was able to get both to show up in `xrandr`.

However when I enable them they either remain blank or show garbage depending on how I arrange them. I'm able to move the mouse off the laptop monitor, and it seems to go somewhere, but the attached monitors don't show any output.

I'm running Debian Buster on a Dell XPS with the Lenovo ThinkPad usb-c dock.

Any suggestions for debugging would be more than welcome!

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-8-amd64

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 2 name:Intel
Provider 1: id: 0x1e8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x1a4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

MiChatz, 2020-05-01:
Hello Julian,    
I just saw your post. I am having a similar setup with some issues.    
My setup is:    
- Dell XPS 13 9350
- Ubuntu 20.04 
- Thinkpad Lenovo Dock Gen2
After displaylink-debian success installation. The system is not detecting the second monitor.    
If you have any suggestions would be highly appreciated.

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 540
VGA: Intel Corporation Iris Graphics 540 (rev 0a)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

nis267, 2020-05-04:
Hello @julianandrews,

- Try this config in your /etc/X11/xorg.conf.d/20-displaylink.conf file and reboot

```
Section "Device"
  Identifier  "DisplayLink"
  Driver      "modesetting"
  Option      "PageFlip" "false"
EndSection
```

Hello @MiChatz,

- In your debug output I see that the /etc/X11/xorg.conf.d/20-displaylink.conf file is not found

- Create it and add the config like above.

julianandrews, 2020-05-04:
@nis267 That fixed the problem for me! Thank you so much!

I'll leave the issue open for now to see if the similar solution works for @MiChatz

MiChatz, 2020-05-04:
Hello,

Thank you for your comments.    
I am afraid that my issue is not so easy to solve.    
I have tried during the last days may different settings for DisplayLink and especially for 20-dislplaylink.config.    
Today I saw @nis267 comment and I tried to reinstall displaylink-debian without success.    
After a while realized that DisplayLink released a new driver for Ubuntu 20.04 (displaylink-debian installation is failing because of that )    
I installed it manually and finally, I have 
```
❯ xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0x151 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x12f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x10d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xeb cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```
I am not sure if everythig is ok with the combination of dell xps13 9350 and Lenovo type-c dock gen2
```
❯ lsusb
Bus 004 Device 004: ID 17ef:a393 Lenovo 
Bus 004 Device 003: ID 17ef:a387 Lenovo USB3.1 Hub
Bus 004 Device 002: ID 17ef:a391 Lenovo USB3.1 Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 17ef:a396 Lenovo 
Bus 003 Device 005: ID 17ef:a38f Lenovo USB2.0 Hub
Bus 003 Device 004: ID 17ef:a395 Lenovo 
Bus 003 Device 003: ID 17ef:a394 Lenovo 
Bus 003 Device 002: ID 17ef:a392 Lenovo USB2.0 Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0c45:670c Microdia Integrated_Webcam_HD
Bus 001 Device 003: ID 04f3:20d0 Elan Microelectronics Corp. Touchscreen
Bus 001 Device 002: ID 0a5c:6412 Broadcom Corp. BCM2045A0
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I am trying to undestand if there is an issue with the dock.    
Any comments would be helpful. thanks in advance

Barabazs, 2020-07-19:
@MiChatz Can you please open a new issue and provide all relevant info you can?

closed 2020-07-19 by Barabazs