AdnanHodzic/displaylink-debian #398: Workaround for evdi patch failure

Erwin-D, 2020-04-29:
I finally found a way to have everything working correctly.
There are some manual steps involved, but here it is:
Start by downloading only these files:
```
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh
```
Then use the following command to comment out the evdi.sh script which we will not use.
`sed -i 's/\.\/evdi.sh/\#\.\/evdi.sh/g' displaylink-debian.sh`

Next step is to download the last evdi release. Modify a small part of the patch. Apply the patch. Use make 
```wget https://github.com/DisplayLink/evdi/archive/v1.6.4.tar.gz
tar -zxvf v1.6.4.tar.gz 
mv evdi-1.6.4/ evdi
cd evdi/
wget https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
sed -i 's/\s\s\s\-\sKVER\=5.2/\+  \- KVER\=5.2/g' evdi-all-in-one-fixes.patch
patchPath="$(pwd)/evdi-all-in-one-fixes.patch"
patch -Np1<$patchPath
sed -E -e 's:SUBDIRS=([^ ]+) :M=\1 &:g' -i 'module/Makefile'
make
cd .. 
```
Now you can continue installing with the slightly modified original script.
```
chmod +x displaylink-debian.sh
sudo ./displaylink-debian.sh --install
```

Erwin-D, 2020-04-29:
And this is my debug information for those that might find it useful.
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.0.0
1.6.4
1.7.0
5.1.26
5.2.14
dev
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: HD
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
Provider 1: id: 0xd2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

KKoovalsky, 2020-04-29:
Hey, I have tried your guide, but unfortunately:
```
$ patch -Np1<$patchPath
patching file .travis.yml
patch: **** malformed patch at line 14: diff --git a/Makefile b/Makefile
```
I will try to debug on my own firstly.

Erwin-D, 2020-04-29:
Did you download the files or cloned the repo?

AdnanHodzic, 2020-04-29:
I would  appreciate it If someone could make a PR out of this.

KKoovalsky, 2020-04-29:
I have cloned the repo and then `git checkout v1.6.4`.

Erwin-D, 2020-04-29:
I have probably messed the sed script...

This way should work:
Replace this part:
`sed -i 's/\s\s\s\-\sKVER\=5.2/\+  \- KVER\=5.2/g' evdi-all-in-one-fixes.patch`
By manually editing the .travis.yml file and adding - KVER=5.2 after the 5.1 entry.

KKoovalsky, 2020-04-29:
The problem is with patching the `Makefile`. I have applied the changes from the patch manually to the `Makefile`. Then I have removed the patching diffs for `.travis.yml` and `Makefile`, so then the diff for `library/Makefile` is first in the `evdi-all-in-one-fixes.patch`. And it all worked fine.

Unfortunately I have some problems to detect my screen. I use USB-C docking station and run Kubuntu 20.04. Need to research more.

persistentpenguin, 2020-05-02:
@KKoovalsky 
Could you please elaborate on how you manually patched the Makefile?
Sorry I'm trying to learn - never patched a makefile before.

If you could explain step-by-step like Erwin did that would be much appreciated. Thanks.

closed 2020-05-03 by Erwin-D