AdnanHodzic/displaylink-debian #399: Ubuntu 20.04 LTS: monitor not detected

vuphan314, 2020-04-29:
The DisplayLink driver installation was successful. But my external ASUS monitor was not detected. What should I try? Thanks!

```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.3.52
1.4.210
4.4.24
5.1.26
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

errst, 2020-04-29:
It didn't work for me either,
`-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0x1e0 cap: 0x2, Sink Output crtcs: 3 outputs: 5 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

4ludwig4, 2020-04-30:
Had the same problem that it didn't detect more than 1 provider.
Was able to solve it by downgrading the evdi version to the last release 1.6.4 instead of using the devel branch being used in the install script. Now it seems to work.

modified evdi.sh
```bash
...
git clone https://github.com/DisplayLink/evdi.git --branch v1.6.4 --single-branch
...
```

Just having problems if I disable the builtin laptop display as the performance decreases a lot this way.

vuphan314, 2020-04-30:
@4ludwig4 Thank you for your comment. Could you describe all steps to make DisplayLink work?

@AdnanHodzic seems to already use [`v1.6.4`](https://github.com/DisplayLink/evdi/tree/v1.6.4) (not branch [`devel`](https://github.com/DisplayLink/evdi/tree/devel)) of EVDI in his script:
https://github.com/AdnanHodzic/displaylink-debian/blob/master/evdi.sh

4ludwig4, 2020-04-30:
@vuphan314 after the modification which is now as a workaround included in the `master` it just worked for me as stated in the readme by running the shell script.

vuphan314, 2020-04-30:
@4ludwig4 Thanks. I tried it.

My external monitor worked for a short while.
```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 1.3.52
1.4.210
4.4.24
5.1.26
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 630 (rev 04)
VGA (3D):
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 1 name:Intel
Provider 1: id: 0x212 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

But the external monitor became undetected again after I logged out or restarted.

nis267, 2020-05-04:
Hello @4ludwig4,
Try this https://github.com/AdnanHodzic/displaylink-debian/issues/293#issuecomment-524834073

nis267, 2020-05-04:
Hello @vuphan314,
Try this https://github.com/AdnanHodzic/displaylink-debian/issues/293#issuecomment-622965579

vuphan314, 2020-05-06:
> Hello @vuphan314,
> Try this [#293 (comment)](https://github.com/AdnanHodzic/displaylink-debian/issues/293#issuecomment-622965579)

@nis267 Thanks! I combined that with the [DisplayLink Ubuntu 5.3 driver](https://www.displaylink.com/downloads/file?d=304), and my ASUS monitor works again! My OS is Ubuntu 20.04.

closed 2020-05-06 by vuphan314